moveit/moveit2 #2076: Use zero-phase filtering in Servo

AndyZe, 2023-04-03:
We should be able to filter forward, then backward to nearly eliminate phase shift. It would be the equivalent of MATLAB `filtfilt`

Shobuj-Paul, 2023-08-06:
> We should be able to filter forward, then backward to nearly eliminate phase shift. It would be the equivalent of MATLAB `filtfilt`

Could you point me towards this code?

AndyZe, 2023-08-06:
Sure. Here's a decent writeup on the concept:

https://community.sw.siemens.com/s/article/butterworth-filter-regular-and-zero-phase

And here's the code:

https://github.com/ros-planning/moveit2/blob/548f5fe3e2f3f83b6de97a7a0abba23ffbca0438/moveit_core/online_signal_smoothing/src/butterworth_filter.cpp#L50

Although reading up on it now, the zero-phase filtering technique only applies to _offline_ filtering. Not online filtering like MoveIt Servo requires. So I guess this issue does not make any sense. Closing it. Sorry @Shobuj-Paul !

closed 2023-08-06 by AndyZe