moveit/moveit2 #2077: Use a "central difference" when calculating velocities in Servo

AndyZe, 2023-04-03:
For lines like this:

https://github.com/ros-planning/moveit2/blob/06e5a0b29107432c8a42e1986d356193383347b8/moveit_ros/moveit_servo/src/servo_calcs.cpp#L783

h/t to @kylc

ibrahiminfinite, 2023-04-04:
Hi @AndyZe ,
By this do you mean to use the actual time difference from the header timestamps ?
Since using `parameters_->publish_period` assumes that the measurements actually arrived within equal intervals ?

AndyZe, 2023-04-04:
The current formula is:

`((future position) - (current position)) / delta_t`

What we should do is:

`(future position) - (previously published position) / delta_t`

For `delta_t` I think it would be safe to use this value since the loop rate is very consistent:  `parameters_->publish_period`

The harder part here is where do you get `previously published position` from? I don't see that it's saved anywhere, currently, so you would have to add a member variable to save it, I think.

ibrahiminfinite, 2023-04-04:
Oh okay.
That was the exact reference I looked up as well !
Got a bit confused there since the  naming was `joint_state` and `original_joint_state`. 

As for the last published position , would it not be possible to use the last point from the last published joint trajectory ?

AndyZe, 2023-04-04:
> As for the last published position , would it not be possible to use the last point from the last published joint trajectory ?
saved in last_sent_command_

Good idea

kylc, 2023-04-04:
Shouldn't the divisor be two time periods, since you are subtracting samples from times `t + dt` and `t - dt` (or `t` and `t - 2 * dt`, depending on your perspective)?

```
joint_state.velocity[i] =
    (joint_state.position.at(i) - last_published_position.at(i)) / (2.0 * parameters_->publish_period);
```

ibrahiminfinite, 2023-04-04:
> Shouldn't the divisor be two time periods, since you are subtracting samples from times `t + dt` and `t - dt` (or `t` and `t - 2 * dt`, depending on your perspective)?

Good catch!
yes, it should divide by  `2dt' here since it is central difference.

AndyZe, 2023-04-04:
I'm not sure about that. `last_published_position` is from just one time period ago

ibrahiminfinite, 2023-04-04:
> Maybe we need to get the command from 2 timesteps in the past.

That would be the first point of  the trajectory in  `last_sent_command_` ?

kylc, 2023-04-04:
@AndyZe For central difference, you need the future command and the previous, at times `t + dt` and `t - dt`. The current one at time `t` is ignored. If you're using the current command and one of either the previous or future, then you're either doing backward or forward difference instead.

I might be lost in the variable names though. You can always verify by comparing the position signal to the integrated velocity signal, which PlotJuggler can do for example.

ibrahiminfinite, 2023-04-04:
At time `t - dt` the joints were at the `first` point of `last_sent_command_` trajectory.
At time `t + dt` the joints will be at `joint_state`.
And the divisor will be ` (t + dt) - (t - dt) = 2 * dt`

AndyZe, 2023-04-04:
Let's say `last_send_command_` was published at `t`.

The current `joint_state` in @ibrahiminfinite's PR #2080 is somewhere between `t` and `t+delta_`.

The next command will be published at `t+delta_t`.

If we're going to use `2*dt` in the denominator (which I think is a good idea) then we need to save 1 earlier command.

AndyZe, 2023-04-04:
I agree with everything Kyle says but we just need to store one more previous command. Does that make sense @ibrahiminfinite or am I missing something?

AndyZe, 2023-04-04:
`last_sent_command_` only contains one point. It doesn't contain the initial state of the robot.

As I see it:

`last_sent_command_` is for the interval `t-dt` to `t`
[this](https://github.com/ros-planning/moveit2/blob/a622feefe4c2221299132538b55160efec245a3b/moveit_ros/moveit_servo/src/servo_calcs.cpp#L775) that you linked in `applyJointUpdate()` is for the interval `t` to t+dt`

The difference is only one timestep

ibrahiminfinite, 2023-04-04:
Ah yes!
You are right, my bad.
I looked at the redundant insertion and mistook it for a full trajectory.

In that case, we need to store one more points.
We can have a `last_commands` vector instead of `last_sent_command`
And keep updating that vector..

ibrahiminfinite, 2023-04-05:
Ended up using an extra `sensor_msgs::msg::JointState`  variable instead, since storing and using the trajectory message would require unnecessary unpacking.

PR #2080 (2023-04-13): Compute velocity using central difference
### Description

This PR changes the calculations in `ServoCalcs` to use the central difference to compute the joint velocities more accurately.

Fixes #2077 

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
closed 2023-04-13 by sjahr