moveit/moveit2 #2079: MoveIt Setup Assistant issues with URDF containing mimic joints

jrgnicho, 2023-04-04:
### Description
The following issues were observed while creating a moveit_config package using the MoveIt Setup Assistant.  The urdf contains mimic joints.
## In the MSA wizard
- 3D display window in wizard was stuck (probably unrelated to mimic joints)

## In the generated moveit_config package
- Added dependency on unavailable `warehouse-ros-mongo` package
- Does not create a ompl_planning.yaml file with prefilled planning configuration info.  Uses default file in moveit_config_utils which does not contain planning group specific planning info.
-  joint_state_broadcaster controller in ros2_controller.yaml isn't properly configured,  it needs a list specifying the actuated joints (exclude mimic joints), otherwise the /controller_manager node crashes and so trajectories can not be executed.
- Needed to add the joint_state_publisher node to publish the state of the mimic joints; the joint_state_publisher node also needs to subscribe to the joint states message published by the joint_state_broadcaster controller, e.g.
```
    joint_state_publisher_node = Node(
        package = 'joint_state_publisher',
        executable = 'joint_state_publisher',
        name = 'joint_state_publisher',
        parameters =[
            {'source_list':['joint_state_broadcaster/joint_states'],
             'rate' : 20.0
                }]
        )
```

### Your environment
* ROS Distro: [Humble]
* OS Version: e.g. Ubuntu 22.04
* Source build
* humble
* Fast DDS

### Steps to reproduce
Generate a moveit config package with the MoveIt Setup Assistant, the robot should contain mimic joints

### Expected behaviour
- 3D display in MSA should be responsive
- The demo.launch file brings up the MotionPlanning panel in Rviz with a planning environment that can plan and execute trajectories
- Trajectory execution should work 

### Actual behaviour
- 3D display in MSA is not responsive
- The demo.launch file crashes
- After fixing the **joint_state_broadcaster** controller trajectory execution does not work, solution requires setting up the joint_state_publisher to provide joint state updates for the mimic joints.

henningkayser, 2023-04-05:
@sea-bass you have been working a lot with mimic joints lately, right? Could you comment if you have experienced similar things, or what needs fixing for the joint state broadcaster to work?

sea-bass, 2023-04-05:
> @sea-bass you have been working a lot with mimic joints lately, right? Could you comment if you have experienced similar things, or what needs fixing for the joint state broadcaster to work?

I've only been consuming pre-made configs with mimic joints, and in fact have never even run MoveIt Setup Assistant before.

However, the [default Panda arm configuration](https://github.com/ros-planning/moveit_resources/tree/ros2/panda_moveit_config/config) that we use everywhere also has a [mimic joint](https://github.com/ros-planning/moveit_resources/blob/ros2/panda_description/urdf/panda.urdf#L222) (one of the gripper fingers), and has a default (`joint_state_broadcaster`)[https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/config/ros2_controllers.yaml] controller, so I'd be surprised if the existence of the mimic joint alone was leading to the problem.

The very last bit about adding a joint state publisher is perhaps something we also have had to do manually.

@jrgnicho Could you perhaps share your configuration, or a minimally reproducible example, that we could take a look at?

jrgnicho, 2023-04-05:
My planning group has 3 revolute joints and 2 prismatic joints where one prismatic joint mimics the preceding prismatic joint.

What I had to change in order for things to work (no crashes and trajectory execution enabled) was the following:
- In the ros2_control.xacro I had to add a tag for all of the actuated joints :
```
        <ros2_control name="${name}" type="system">
            <hardware>
                <!-- By default, set up controllers for simulation. This won't work on real hardware -->
                <plugin>mock_components/GenericSystem</plugin>
            </hardware>
            <joint name="actuated_joint_0">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['actuated_joint_0']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>
            <joint name="actuated_joint_1">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['actuated_joint_1']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>
            ...
            ...
    
        </ros2_control>
```
- Then in the ros2_controller.yaml I had to specify the list of actuated joints that the joint state broadcaster needed to monitor, I did this as follows:
```
joint_state_broadcaster:
   ros__parameters:
      use_local_topics: True
      joints:
         - actuated_joint_0
         - actuated_joint_1
         - actuated_joint_2
         - actuated_joint_prismatic_1
         - gripper_actuated_joint_1
         - other_actuated_joint
         - .....
      interfaces:
         - position
         - velocity
```
All the joints in this list must have an entry in the ros2_control.xacro  file.

- Lastly I added a joint state publisher node to the move_group.launch.py file
```
    joint_state_publisher_node = Node(
        package = 'joint_state_publisher',
        executable = 'joint_state_publisher',
        name = 'joint_state_publisher',
        parameters =[
            {'source_list':['joint_state_broadcaster/joint_states'],
             'rate' : 20.0
                }]
        )
```

This is the info that I can share on this configuration, hope this helps.

swiz23, 2023-04-09:
I have also run into the issue where MSA does not generate an ompl_planning.yaml file. I'd be interested to see this resolved as well.

sea-bass, 2023-04-11:
OK, I started taking a look. A few updates:

* I think the `ompl_planning.yaml` template actually hasn't been ported over from MoveIt 1 -- so it's something that still needs to be done for MoveIt 2
* The `warehouse_ros_mongo` package indeed doesn't build for ROS 2 yet, but it can be removed in the "Launch Files" section of MSA by unchecking "Warehouse DB Launch"

Will next look at the mimic joints issues next. My guess, though, is that this is also some new functionality that needs to be supported.

jrgnicho, 2023-04-17:
@sea-bass 

1 - ros2 control setup
That's interesting that it worked for you by adding the joint tags for the mimic joints, in my setup I added only the joints that are fully actuated but not the mimic ones.  

3 - OMPL yaml file
In my case I ensured that the "default_planner_request_adapters/AddTimeParameterization" adapter was in the list of adapters.  I imagine that as long as there's an adapter that can fill in the trajectory time stamp fields then trajectory execution should just work.

I also manually added the per "planning group" planner_configs for each planning group in my moveit config package.  So the end of my ompl_planning.yaml file looks more or less like this:
```
main_arm:
  default_planner_config: RRTConnect
  planner_configs:
    - AnytimePathShortening
    - SBL
    - EST
    - LBKPIECE
    - BKPIECE
    - KPIECE
    - RRT
    - RRTConnect
    - RRTstar
    - TRRT
    - PRM
    - PRMstar
    - FMT
    - BFMT
    - PDST
    - STRIDE
    - BiTRRT
    - LBTRRT
    - BiEST
    - ProjEST
    - LazyPRM
    - LazyPRMstar
    - SPARS
    - SPARStwo
  projection_evaluator: joints(arm_joint_0,arm_joint_1)
  longest_valid_segment_fraction: 0.001
```

This allows for the MoveIt Planning Panel in Rviz to populate the bottom OMPL dropdown in the "Context" tab with the list of available planners.

> As for the UI being unresponsive, do you mean that trying to zoom / pan / rotate in the display on the right does nothing? Because I too see that on Rolling right now.

Yes that is exactly what I observed but in humble on Ubuntu 22.04

> Additionally, if I press "Ctrl" while MSA is up, I get a stack trace:

I did not know that there was a function tied to the "Ctrl" key so I did not try that.  I did not experience any crashes however while using the MSA.

closed 2024-02-27 by github-actions[bot]