moveit/moveit2 #2083: moveit2 crash ild- ros2 foxy ub

galversano, 2023-04-05:
### Description

i try to clone and build moveit2 for foxy but every the time im build the computer stuck in the bulild of moveit_ros_visualization

### Your environment
* ROS Distro: [Foxy|
* OS Version: e.g. Ubuntu 2004
* Source or Binary build?source

### Steps to reproduce

 i follow this guide here: https://moveit.ros.org/install-moveit2/source/

i run this command:
`git clone https://github.com/ros-planning/moveit2.git -b $ROS_DISTRO
for repo in moveit2/moveit2.repos $(f="moveit2/moveit2_$ROS_DISTRO.repos"; test -r $f && echo $f); do vcs import < "$repo"; done
rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`

and after this i try to build with this command:
`cd $COLCON_WS
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`

### Actual behaviour
 when the system try to build the pkg of moveit_ros_visualization the computer crash and i need to restart the computer.
i try to use this command:
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 1
`
but its is still crash
anyone can help me?

JafarAbdi, 2023-04-12:
Please make sure to increase your swap memory and test again, feel free to re-open if it still doesn't help!

closed 2023-04-12 by JafarAbdi

AndyZe, 2023-04-13:
Here's another tip on managing RAM automatically:

`sudo apt-get install zram-config`
This enables compression in your RAM, and should prevent system lockups some people experience when compiling very memory hungry packages from source (at the expense of some CPU resources for the compression).