moveit/moveit2 #2100: Use of `make_shared_from_pool` in MoveIt Servo

ibrahiminfinite, 2023-04-10:
There is a `make_shared_from_pool.h` header in MoveIt Servo package,  the indented use appears to be efficient creation of messages objects that are published from within servo, however , while it is included in most of the other files, presently it only used in `pose_tracking.cpp`.

Having the required message type as a member variable and reusing it might be a good alternative.
Unless there is a specific reason not to go that way.

ibrahiminfinite, 2023-04-27:
@AndyZe 
This is still left.
Currently used in `pose_tracking.cpp` , [here](https://github.com/ros-planning/moveit2/blob/dc7643502adbeaa41dfd370605460cfd45f26b7f/moveit_ros/moveit_servo/src/pose_tracking.cpp#L279) and [here](https://github.com/ros-planning/moveit2/blob/dc7643502adbeaa41dfd370605460cfd45f26b7f/moveit_ros/moveit_servo/src/pose_tracking.cpp#L330).

Should this be applied  to other parts or do we want to remove this utility from the codebase ?

AndyZe, 2023-04-27:
let's ask @tylerjw

ibrahiminfinite, 2023-06-12:
Closing this since it is taken care of in the new refactor.
And doesn't need to be addressed in the current version.

closed 2023-06-12 by ibrahiminfinite