moveit/moveit2 #2101: Current position not relative to reference coordinate system after setting with move_group.setPoseReferenceFrame()

yeyanlei, 2023-04-11:
### Description

Hello everyone,

I'm currently working on a project that involves setting a reference coordinate system for the robot. I have set the reference coordinate system using the following line of code:

move_group.setPoseReferenceFrame("panda_link0");

However, when I try to get the current position of the robot, it seems not to be relative to the reference coordinate system that I set. Here's the code I'm using to get the current position:

  geometry_msgs::msg::PoseStamped current_pose = move_group.getCurrentPose();

  std::cout << "Current end-effector position: " << std::endl;
  std::cout << "x: " << current_pose.pose.position.x << std::endl;
  std::cout << "y: " << current_pose.pose.position.y << std::endl;
  std::cout << "z: " << current_pose.pose.position.z << std::endl;

log:
[move_group_interface_tutorial-1] Current end-effector position: 
[move_group_interface_tutorial-1] x: 0.30702
[move_group_interface_tutorial-1] y: -5.22133e-12
[move_group_interface_tutorial-1] z: 1.59027

expected output
[move_group_interface_tutorial-1] Current end-effector position: 
[move_group_interface_tutorial-1] x: 0.30702
[move_group_interface_tutorial-1] y: -5.22133e-12
[move_group_interface_tutorial-1] z: 0.59027
urdf:
<link name="world"/>
    <joint name="joint0" type="fixed">
        <parent link="world" />
        <child link="panda_link0" />
        <origin rpy="0 0 0" xyz="0 0 1" />
    </joint>
    <link name="panda_link0">

### Your environment
* ROS Distro: [Humble]
* OS Version:Ubuntu 22.04
* Binary build

yeyanlei, 2023-04-11:
urdf: 
<link name="world"/>
    <joint name="joint0" type="fixed">
        <parent link="world" />
        <child link="panda_link0" />
        <origin rpy="0 0 0" xyz="0 0 1" />
    </joint>
    <link name="panda_link0">

yeyanlei, 2023-04-11:
@rhaschke @henningkayser @AndyZe

rhaschke, 2023-04-11:
According to the docs, `setPoseReferenceFrame()` is only used to specify the reference frame for _targets w/o a frame specified_:
https://github.com/ros-planning/moveit2/blob/25500e6120a4ba8acab39bd6e6d344683fa89680/moveit_ros/planning_interface/move_group_interface/include/moveit/move_group_interface/move_group_interface.h#L642-L643

`getCurrentPose()` on the other hand returns a `PoseStamped`, thus explicitly specifying the reference frame (which is the root link of the robot). If you want to express the pose in a different frame, please resort to TF or other tools.
https://github.com/ros-planning/moveit2/blob/25500e6120a4ba8acab39bd6e6d344683fa89680/moveit_ros/planning_interface/move_group_interface/include/moveit/move_group_interface/move_group_interface.h#L928

closed 2023-04-11 by rhaschke

yeyanlei, 2023-04-12:
@rhaschke Thank you for your response!
Based on the information you provided, I understand that setPoseReferenceFrame() is only used to specify the reference frame for targets without a specified reference frame. On the other hand, getCurrentPose() returns a PoseStamped object with an explicitly specified reference frame, which is the root link of the robot. If I want to express the pose in a different frame, I should resort to using TF or other tools.

Your response has helped me gain a clearer understanding of the issue. I truly appreciate your assistance! Have a great day!