moveit/moveit2 #2103: Constraint Planning Segfault

MarqRazz, 2023-04-12:
### Description

I have been working on a problem where I need orientation constraints to be held when transporting an object and have been observing segmentation faults in MoveIt. In my setup I'm also trying to build databases to speed up planning but when I enable [multi_query_planning_enabled](https://moveit.picknik.ai/main/doc/examples/ompl_interface/ompl_interface_tutorial.html?highlight=multi_query_planning_enabled#persistent-roadmaps) motion planning crashes.

### Your environment
* ROS Distro: [Humble & Rolling]
* OS Version: e.g. Ubuntu 22.04
* Source build
* Cyclone DDS

### Steps to reproduce
I have made a working [docker container with instructions](https://github.com/MarqRazz/moveit2_tutorials/tree/pr-constraint_segfault_example#debug-constraint-planning-segmentation-fault) that includes some details on where I have been seeing the faults. In this container I have modified [ompl_constrained_planning_tutorial](https://github.com/MarqRazz/moveit2_tutorials/blob/pr-constraint_segfault_example/doc/how_to_guides/using_ompl_constrained_planning/src/ompl_constrained_planning_tutorial.cpp) to include some collision objects around the robot and to attempt to find motion plans forever because the fault happens on the second or third attempt. I'm asking for multiple plans because in my actual application I'm using MTC and this happens when my constraint stage has a parent with multiple solutions.

### Expected behaviour
I should be able to run the modified tutorial example without MoveIt crashing.

### Actual behaviour
Program crashes with a segfault.

### Backtrace or Console output

[segfault backtrace](https://gist.github.com/MarqRazz/4fa972f736746913c105f6be899e1cad)

henningkayser, 2023-04-26:
@MarqRazz I think I've found what is broken in OMPL. However, I don't have a great suggestion for a fix but maybe a better workaround.

multi-query planners don't support being run in multiple threads that share the same planner data (I don't recall why). The answer to [your question here](https://github.com/MarqRazz/moveit2_tutorials/blob/b11b8a7ca404fc12c458495ecf785d7934855ce1/doc/how_to_guides/using_ompl_constrained_planning/src/ompl_constrained_planning_tutorial.cpp#L120) is that planning attempts directly translate to planning threads internally (all attempts at the same time) which OMPL implements in the [solve](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L847) function. Do you see the tiny comment in [this line](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L855)? I think that's the problem. Please try setting planning attempts to 1. If that works, we should enforce this setting for multi-query planners.

MarqRazz, 2023-04-26:
> Please try setting planning attempts to 1. If that works, we should enforce this setting for multi-query planners.

No that does not fix the problem. Sorry I left that line in the example but as you noted the tiny comment in [this line](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L855) forces it to solve it a single thread.

> Edit: Parallel plan doesn't work since it's clearing the planner instance and creating new ones with the same config.

The problem is that it's trying to reuse the same planning object between requests when the multi_query_planning_enabled flag is enable.

henningkayser, 2023-04-26:
> > Please try setting planning attempts to 1. If that works, we should enforce this setting for multi-query planners.
> 
> No that does not fix the problem. Sorry I left that line in the example but as you noted the `if` statement with the tiny comment in [this line](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L855) forces it to solve in a single thread.

Well, you're obviously right with that.
 
> > Edit: Parallel plan doesn't work since it's clearing the planner instance and creating new ones with the same config.
> 
> The problem is that it's trying to reuse the same planning object between requests when the multi_query_planning_enabled flag is enable.

That's the purpose of that flag, no? The problem is that the instance is being wiped somewhere which to me means that there is something specific to how planner instances are handled by the context. I'll have to dig a bit deeper for that, though. Reusing the instance should work. If not, storing and reusing the planner data only should be implemented instead.

MarqRazz, 2023-04-26:
Andy and I think it might have to do with the robot_state that is used to construct the objects inside the planner (see [this](https://github.com/ros-planning/moveit2/pull/2130#discussion_r1176674229) comment).

henningkayser, 2023-05-23:
Allowing to [cache constrained planning contexts](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L397) fixes the segfault for me. Obviously, this is not a good solution for documented reasons.

henningkayser, 2023-05-23:
ok, I think I've figured it out...

With constrained planning, [we are not caching planning contexts](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L397). The reason is that each request can have a different mix of constraints which would require building a new [StateValidityChecker](https://ompl.kavrakilab.org/classompl_1_1base_1_1StateValidityChecker.html) which is in turn owned by the [StateInformation](https://ompl.kavrakilab.org/classompl_1_1base_1_1SpaceInformation.html). Planners keep the StateInformation as const member. Problem is, the StateValidityChecker has a [raw pointer](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/state_validity_checker.h#L90) to the PlanningContext [it was created with](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L122), and eventually runs into a segfault when that one goes out of scope. Multi-query-planning with the same planner instance would basically depend on caching planning contexts which we don't do for the described reason.

There are different ways to approach this:
1. Simply recreate planner instances with each call and live with the performance implications (current solution), possibly find out ways to reduce the time overhead
2. Update the members of StateValidityChecker and ProblemDefinition of the cached planner instance (there is [some API](https://ompl.kavrakilab.org/classompl_1_1base_1_1Planner.html) for this), hoping that this is sufficient (could work, needs to be investigated, maybe not worth the effort)
3. Implement PlanningContext caching for constrained planning, requiring us to basically parse and keep track of constraint groups
4. Expose an API for updating the SpaceInformation etc of a planner

However, if we have to construct and configure constrained planning requests anew each time anyway, there is probably not a great reason to keep the planner instance alive. Caching SpaceInformation, StateValidityChecker etc (basically the PlanningContext) is really what speeds up multi-query planning, so option 3 would be the only proper fix to me.

A short improvement for now could be to simply disable multi-query-planning for all constrained planning requests, but still allow it for others.

radhen, 2024-05-20:
@henningkayser  @MarqRazz @sjahr  are there any updates on this issue? Any concrete solution that has worked for you? 

I'm having a similar problem with my setup. I'm using OMPL, MoveIt, and ROS Humble on Ubuntu 22.04. I'm trying to save a roadmap using the PersistentLazyPRMstar planner in a constraint setting, so I can use it later for planning. In my setup, I use different planners for different types of motion.

I'm new to digging into the MoveIt and OMPL code, but I think I kinda understand what's going wrong, based on what's been discussed here and the call stack that is linked up. But I need help putting one of @henningkayser's suggestions into action.

Here's the link to the text file ([ompl_moveit_seg_fault_call_stack.txt](https://github.com/moveit/moveit2/files/15371684/ompl_moveit_seg_fault_call_stack.txt)) with the call stack for the segmentation fault I'm getting. Any assistance or insights you could offer would be greatly appreciated. Thank you in advance for your help!

radhen, 2024-05-23:
The fix mentioned in the issue #2130 helped we clear the segmentation fault. Call trace linked in the above comment. Here is the link to the [Files Changed](https://github.com/moveit/moveit2/pull/2130/files).

radhen, 2024-05-24:
Hello @MarqRazz. Thank for the response. My setup also consists of a 6DOF robot mounted to a linear rail. I definitely saw an improvement in the planning time (reduced almost by a factor of 10) once the roadmap was build. I built the roadmap using the PresistentLazyPRMstar. Once it was built, I was using it with PresistentLazyPRM to test. Note my PRM goes through a constraint pipeline. The success rate rather saw a decrement, but I believe if I built a more denser roadmap then the success rate should go up. To summarize, the PRM works well, when it does, and brings the planning time considerably down. Please let me know if you have any other questions.