moveit/moveit2 #2104: publish_scene_from_text does not work without --scene

christian-rauch, 2023-04-13:
### Description

The tool [`moveit_publish_scene_from_text`](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/planning_components_tools/src/publish_scene_from_text.cpp) does not work without providing `--scene` and hence cannot publish on topic `/planning_scene_world`.

### Origin of the Problem

The `--scene` flag changes the topic of where the scene is published on. This is controlled by `full_scene` in the code:
https://github.com/ros-planning/moveit2/blob/c686bb8dc312029430d6c353799ef462d05bb3e7/moveit_ros/planning/planning_components_tools/src/publish_scene_from_text.cpp#L62-L67
and then sets the topic:
https://github.com/ros-planning/moveit2/blob/c686bb8dc312029430d6c353799ef462d05bb3e7/moveit_ros/planning/planning_components_tools/src/publish_scene_from_text.cpp#L77-L86

The topic is either `/planning_scene` (with `--scene`) or `/planning_scene_world` (without `--scene`).

However, when the node waits here:
https://github.com/ros-planning/moveit2/blob/c686bb8dc312029430d6c353799ef462d05bb3e7/moveit_ros/planning/planning_components_tools/src/publish_scene_from_text.cpp#L103-L105
it only considers the option for the `pub_scene` topic which is only defined when `--scene` is set.

Removing the last "waiting" part fixes the issue and successfully publishes the "world" on `/planning_scene_world`.

### Steps to reproduce
Terminal 1 (publish scene):
```sh
ros2 run moveit_ros_planning moveit_publish_scene_from_text ${scene_file} --ros-args -p robot_description:="${urdf}" -p robot_description_semantic:="${srdf}"
```

Terminal 2 (monitor scene):
```
ros2 topic echo /planning_scene_world
```

### Expected behaviour
The scene message should be published on topic `/planning_scene_world`.

### Actual behaviour
No messages arrive.

henningkayser, 2023-12-05:
@christian-rauch what's the current status on this issue? Have you found a suitable solution like defaulting to "--scene" input? We are currently not using these tools and would long-term consider rewriting convenience features like this in Python. If you provide a PR, we are happy to assist.

LetianLi, 2025-06-25:
This bug should be pretty simple to fix by just renaming the `pub_scene_world` variable back to `pub_scene` as it looks like at some point it was modified between moveit1 and moveit2. 

The moveit1 version does not have this issue from my experience, see source at https://github.com/moveit/moveit/blob/e6cdf5114d5a2313c15108d40d9aa4e605dbb434/moveit_ros/planning/planning_components_tools/src/publish_scene_from_text.cpp#L66

Here's the diff between the two: https://www.diffchecker.com/OULyIl9h/