moveit/moveit2 #2107: Port move_group tests to ROS 2

sjahr, 2023-04-13:
move_group still contains [integration tests](https://github.com/ros-planning/moveit2/tree/main/moveit_ros/move_group/test) that are not ported to ROS 2

Shobuj-Paul, 2023-04-13:
I shall take this up. Have already started work on this.

Shobuj-Paul, 2023-08-09:
This particular issue slipped my mind, I'll restart work on this.

Shobuj-Paul, 2023-08-30:
Hi,

I had a few doubts regarding this porting acitivity. Should I write the tests in C++ or python, since the move_group package is inherently a C++ package. Till now I have been trying to port it as a python node (which has been slow). This was not an issue for moveit since in ROS1 there was no difference between types of packages.