moveit/moveit2 #2127: Constant zero-velocity commands in Servo breaks MoveIt Studio

sea-bass, 2023-04-20:
The changes to Servo merged in https://github.com/ros-planning/moveit2/pull/1904 broke MoveIt Studio for 2 different reasons, so we had to revert it in our fork. Wanted to make an issue to track this, as the current state will prevent us from moving to later versions.

## Issues
### Gazebo Drift
Because of the issues described in https://github.com/ros-planning/moveit2/issues/1857, when we have a Gazebo simulation this constant drift will actually make the arm sag under its own weight forever once any single Servo command is published to the robot. So having the option to stop Servo messages is desirable as a workaround.

### Switching between Cartesian and Joint Commands
MoveIt Studio has two separate screens for publishing Cartesian vs. Joint commands. Because of this behavior, once we publish one set of commands, Servo will push them out infinitely, which effectively nullifies the other set of commands forever. Meaning we can't switch between these modes at runtime.

## Possible Solutions
1. Revert the original PR, keeping around the "gradual halt" behavior around if we want
2. Make this behavior configurable via a parameter

AndyZe, 2023-04-20:
I'd rather not revert the MoveIt2 PR https://github.com/ros-planning/moveit2/pull/1904 b/c it was a nice 100-line cleanup and it fixed a different issue. Could you guys make use of the Servo `pause` and `unpause` services?

https://github.com/ros-planning/moveit2/blob/a1fce2d077f16673d8dd540f5b1072b5bc0e56f5/moveit_ros/moveit_servo/src/servo_node.cpp#L66

sea-bass, 2023-04-20:
It seems like we are actually calling the services already, so we'll have to dig into what's going wrong...

ibrahiminfinite, 2023-04-20:
> MoveIt Studio has two separate screens for publishing Cartesian vs. Joint commands. Because of this behavior, once we publish one set of commands, Servo will push them out infinitely, which [...]

What does "two separate screens" mean, is this like 2 separate terminals running 2 separate servo instances or 2 separate publishers ?

AndyZe, 2023-04-20:
^ good question.

I just remembered (from testing) that if you set Servo to paused, it publishes 0-velocity commands continuously. What @sea-bass really ought to do, most likely, is call the start/stop services:

`ros2 service call /servo_node/stop_servo std_srvs/srv/Trigger {}`

AndyZe, 2023-04-20:
I just tested start/stop, even with a change in robot position in-between. It seems fine.

sea-bass, 2023-04-20:
> What does "two separate screens" mean, is this like 2 separate terminals running 2 separate publishers ?

It means we have 2 publishers to each of these topics: https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/src/servo_calcs.cpp#L115-L122

sea-bass, 2023-04-20:
> I just remembered (from testing) that if you set Servo to paused, it publishes 0-velocity commands continuously. What @sea-bass really ought to do, most likely, is call the start/stop services:
> 
> `ros2 service call /servo_node/stop_servo std_srvs/srv/Trigger {}`

Will try this -- thanks, Andy!

sea-bass, 2023-04-21:
Gave it some more thought. I think tapering commands down after X time of inactivity should continue being the role of the servo node, and going off removing lines of code at the main rationale may not be the ideal solution here.

I think we should look to fix this issue. Not necessarily by blanket reverting, but by finding something clean that works for this and the other issues the cleanup intended to address.

Thoughts?

AndyZe, 2023-04-24:
> I think tapering outgoing commands down after X time of incoming command inactivity should continue being the role of the servo node

This does happen! Relevant code [here](https://github.com/ros-planning/moveit2/blob/dd00f075ffc9b18b6311dc25102d3050cc9d6102/moveit_ros/moveit_servo/src/servo_calcs.cpp#L462):

```
  // Skip servoing publication if both types of commands are stale.
  if (twist_command_is_stale_ && joint_command_is_stale_)
  {
    rclcpp::Clock& clock = *node_->get_clock();
    RCLCPP_DEBUG_STREAM_THROTTLE(LOGGER, clock, ROS_LOG_THROTTLE_PERIOD,
                                 "Skipping publishing because incoming commands are stale.");
    filteredHalt(*joint_trajectory);
  }

  if (!paused_)
  {
    ... continue normally
```

Maybe you mean, the Servo node should stop publishing completely after the robot halts?

closed 2023-04-24 by sea-bass