moveit/moveit2 #2131: Replace the nullpointer checking in the Planning Scene with std::optional

AndyZe, 2023-04-21:
Stuff like [this](https://github.com/ros-planning/moveit2/blob/acbf3770763e58e4f86ce0dea251a50cd4987fe6/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h#L1035):

`std::unique_ptr<ObjectTypeMap> object_types_;`

```
  if (!object_types_)
  {
    ObjectTypeMap kc;
    parent_->getKnownObjectTypes(kc);
    object_types_ = ObjectTypeMap(kc);
  }
  else
  {
    ...
```

Ought to be replaced with std::optional<ObjectTypeMap>.

SuyashSapkal, 2023-04-22:
Hi @AndyZe , @tylerjw and @henningkayser 

I would like to work on this. Let me know what things I should be working on.

AndyZe, 2023-04-23:
I think we need to at least wait for a review of #2132 first. If it's acceptable, you could apply the same techniques again.

Thanks @SuyashSapkal

MEHARRAJ, 2023-08-29:
I would like to work on this. Let me know what things I should be working on.

sjahr, 2023-09-04:
@MEHARRAJ Thank you, feel free to open a PR with the necessary changes

zearocode, 2023-10-17:
Hi @AndyZe @tylerjw , I would like to work on this issue . May I know what changes should I do ?

utkarshanand140, 2024-02-22:
Hey @AndyZe @tylerjw , has there been any progress on this ? If not, I would like to start work on this. Kindly lemme know.

PR #2454 (2024-05-01): Use std::optional instead of nullptr checking
### Description

Wrapped scene_transforms_, robot_state_, acm_ and object_types_ in std::optional to handle missing data instead of checking for nullptr.
Fixes #2131 

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/ros-planning/moveit2/blob/main/MIGRATION.md) notes
- [... cut]
closed 2024-05-01 by EzraBrooks