moveit/moveit2 #2134: Add documentation for setting up moveit2 docker containers

abhijelly, 2023-04-23:
**Is your feature request related to a problem? Please describe.**
There is no documentation for  setting up a moveit2 docker container.

**Describe the solution you'd like**
I propose to add documentation for setting moveit2 docker container at least for development purpose inside `.docker` folder.

**Describe alternatives you've considered**
I've searched for the docs inside `.docker` folder and on moveit documentation but did not find anything like ros-controls/ros2_control has created inside their `.docker` folder [here](https://github.com/ros-controls/ros2_control/tree/master/.docker) .

JafarAbdi, 2023-04-23:
We have instructions on the tutorials' website https://moveit.picknik.ai/humble/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html?highlight=docker, have you tried it?

abhijelly, 2023-04-25:
> We have instructions on the tutorials' website https://moveit.picknik.ai/humble/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html?highlight=docker, have you tried it?

I have not. Thanks for sharing it!

closed 2023-04-25 by abhijelly