moveit/moveit2 #2136: Jerky motion with MoveIt Servo

Pomorondza, 2023-04-24:
### Description

Jerky motion occurs when using MoveIt Servo to control the robot (UR3e) in Cartesian space. The controller I am using is scaled_joint_trajectory_controller (position+velocity command interfaces). What I observe when looking at the /scaled_joint_trajectory_controller/states topic is that the desired joint positions and velocities are within the limits, however desired joint accelerations overshoot (abnormal values). What I think it happens is that sporadically the robot velocity drops to zero, causing the acceleration to increase with the next commanded velocity. This phenomena gets worse as the servoing frequency increases. This results in jerky motion of the robot, however in the correct direction. Eventually, the robot reaches the target, yet due to the jerkiness, the velocity is impossible to control.

I am desperately trying to solve this issue, and still I have no clue. I have tried the following:
* Switching to a RT kernel
* Decreasing the servoing frequency
* Removing joint limits

### Environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04 RT kernel

### Expected behaviour
The motion of the robot should be smooth and executed in the commanded speed.

### Actual behaviour
The motion of the robot is jerky and the execution speed is unpredictable and uncontrollable.

ibrahiminfinite, 2023-04-24:
Hi,
Can you specify which commit you are on  or your moveit2 version.
This issue has been popping up lately , and efforts are going on to isolate it

ibrahiminfinite, 2023-04-24:
If possible can you also check if an older version of moveit2 works for you without the jerking issue.

Pomorondza, 2023-04-24:
Hi, thank you for your response. I updated the version. Do you have an idea which older version I could try?

ibrahiminfinite, 2023-04-24:
> Hi, thank you for your response. I updated the version. Do you have an idea which older version I could try?

Would you be able to try versions
- 2.5.0
- 2.6.0
- 2.7.0 

And check if the issue is present in all 3 or if it is only in a particular version.

gavanderhoorn, 2023-04-24:
> The controller I am using is scaled_joint_trajectory_controller [..]

is that even supported by Servo?

I was under the impression only `JointGroup*Controller`s could be used, which would make sense: Servo doesn't generate trajectories. It basically generates position/velocity setpoints.

Pomorondza, 2023-04-24:
Hi, thank you for your reply.
> is that even supported by Servo?

You are right, and it does not make complete sense to send _single point trajectories_ to the controller that is meant for that, as I am doing now through Servo. However, this is supported in Servo [settings ](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/config/panda_simulated_config.yaml):

> command_out_type: trajectory_msgs/JointTrajectory

The reason why I haven't switched to a more reasonable controller, such as the one you suggest, is that it is not yet supported in the ROS2 UR driver, or I am missing something.

ibrahiminfinite, 2023-04-24:
> However, this is supported in Servo [settings ](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/config/panda_simulated_config.yaml):

This setting only specifies the  `type` , the trajectory being send will however only have a single point.

gavanderhoorn, 2023-04-24:
> The reason why I haven't switched to a more reasonable controller [..] is that it is not yet supported in the ROS2 UR driver

what is not supported?

The UR driver will load whatever `ros2_control` supported controller you tell it to.

As long as the required state and command interfaces are offered by the driver, it will load it.

And the `JointGroup*Controller`s are just about the simplest 'controllers' there are (most are forwarding controllers, so they don't really implement any control law at all).

From your description it sounds like what you observe is caused by the `JointTrajectoryController` doing what it's supposed to do: attempt to smoothly interpolate between the many, single point, trajectories you are sending it at a high frequency.

Pomorondza, 2023-04-24:
Aha! Okay, thank you. I will try it out with another controller, and let you know of the outcome.

andrey-pr, 2023-04-28:
I tested a `JointGroupPositionController` and fixed my similar issue

Pomorondza, 2023-05-01:
> I tested a `JointGroupPositionController` and fixed my similar issue

Hi, can you tell me please what are your environment settings (ROS and MoveIt versions)?

Update on my side:

1. I have tried switching the controller, and now, after also adapting the [ros_contro.urscript](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/foxy/ur_robot_driver/resources/ros_control.urscript), the jerkiness was gone, but another issues occurred. The issue is that the robot joints "drift" when trying to control even a single joint of the robot (#1857).
2. I haven't tried switching to another version of MoveIt yet, seems like I will have to install ROS2 Humble/Rolling first.

andrey-pr, 2023-05-01:
> > I tested a `JointGroupPositionController` and fixed my similar issue
> 
> Hi, can you please tell me what are your environment settings (ROS and MoveIt versions)?
> 
> 
> Update on my side:
> 
> 1. I have tried switching the controller, and now, after also adapting the [ros_contro.urscript](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/foxy/ur_robot_driver/resources/ros_control.urscript), the jerkiness was gone, but another issue occurred. The issue is that the robot joints "drift" when trying to control even a single joint of the robot (#1857).
> 2. I haven't tried switching to another version of MoveIt yet, seems like I will have to install ROS2 Humble/Rolling first. 

I used ROS 2 Humble and MoveIt 2.5.4 and 2.7

Pomorondza, 2023-05-04:
My problem was indeed fixed when switching to JointGroupPositionController and updating to ROS2 Humble and Moveit 2.5.4. Thank you everyone.

closed 2023-05-04 by Pomorondza

maxdgaa, 2023-05-27:
Hi, 
Might be a dumb question but how are you using a JointGroupPositionController when servo only supports std_msgs/Float64MultiArray or trajectory_msgs/JointTrajectory?

gavanderhoorn, 2023-05-27:
Because [`JointGroupPositionController` is a `ForwardCommandController`](https://github.com/ros-controls/ros2_controllers/blob/45d008321fe593db0e7f68921baa326b25c38daa/position_controllers/include/position_controllers/joint_group_position_controller.hpp#L36) which [accepts `std_msgs/Float64MultiArray`](https://github.com/ros-controls/ros2_controllers/blob/45d008321fe593db0e7f68921baa326b25c38daa/forward_command_controller/include/forward_command_controller/forward_controllers_base.hpp#L32).

maxdgaa, 2023-05-27:
I see, thank you.

JaisonJose241, 2023-08-20:
The problem is only due to execution of each controller, the trajectory controller does a trapezoidal approach to reach a point while the position controllers don't.