moveit/moveit2 #2141: Moveit Servo Crashes

robertokcanale, 2023-04-26:
After I have the robot running and the servo node running, I want to to start using the servo node. I call the service with: 

`ros2 service call  /servo_node/start_servo std_srvs/srv/Trigger {} `
However, I call the Service to activate the servo, the node crashes. Here is the log. 

[servo_node_main-3] servo_node_main: pthread_mutex_lock.c:438: __pthread_mutex_lock_full: Assertion `e != ESRCH || !robust' failed.
[servo_node_main-3] Stack trace (most recent call last):
[servo_node_main-3] #21   Object "", at 0xffffffffffffffff, in 
[servo_node_main-3] #20   Object "/opt/ros/humble/lib/moveit_servo/servo_node_main", at 0x55fcd21abca4, in _start
[servo_node_main-3] #19   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7f7bf44aae3f]
[servo_node_main-3] #18   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7f7bf44aad8f]
[servo_node_main-3] #17   Source "./src/servo_node_main.cpp", line 50, in main [0x55fcd21abb4d]
[servo_node_main-3] #16   Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f7bf49f9154, in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>)
[servo_node_main-3] #15   Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f7bf49f8f3f, in rclcpp::executors::SingleThreadedExecutor::spin()
[servo_node_main-3] #14   Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f7bf49f16e5, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[servo_node_main-3] #13   Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f7bf49f1379, in rclcpp::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>)
[servo_node_main-3] #12   Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f7bf49f3965, in 
[servo_node_main-3] #11   Object "/home/asr/Desktop/CartinLFD/install/ur10eservo/lib/libservo_node.so", at 0x7f7bf4e5d783, in rclcpp::Service<std_srvs::srv::Trigger>::handle_request(std::shared_ptr<rmw_request_id_s>, std::shared_ptr<void>)
[servo_node_main-3] #10   Object "/home/asr/Desktop/CartinLFD/install/ur10eservo/lib/libservo_node.so", at 0x7f7bf4e5d190, in rclcpp::AnyServiceCallback<std_srvs::srv::Trigger>::dispatch(std::shared_ptr<rclcpp::Service<std_srvs::srv::Trigger> > const&, std::shared_ptr<rmw_request_id_s> const&, std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > >)
[servo_node_main-3] #9    Object "/home/asr/Desktop/CartinLFD/install/ur10eservo/lib/libservo_node.so", at 0x7f7bf4e51230, in moveit_servo::ServoNode::startCB(std::shared_ptr<std_srvs::srv::Trigger_Request_<std::allocator<void> > > const&, std::shared_ptr<std_srvs::srv::Trigger_Response_<std::allocator<void> > > const&)
[servo_node_main-3] #8    Source "./src/servo.cpp", line 72, in start [0x7f7bf4d7ab7a]
[servo_node_main-3] #7    Source "./src/servo_calcs.cpp", line 244, in start [0x7f7bf4d7dd7f]
[servo_node_main-3] #6  | Source "./src/servo_calcs.cpp", line 296, in lock_guard
[servo_node_main-3]     | Source "/usr/include/c++/11/bits/std_mutex.h", line 229, in lock
[servo_node_main-3]     |   228:       explicit lock_guard(mutex_type& __m) : _M_device(__m)
[servo_node_main-3]     | > 229:       { _M_device.lock(); }
[servo_node_main-3]     |   230: 
[servo_node_main-3]     |   231:       lock_guard(mutex_type& __m, adopt_lock_t) noexcept : _M_device(__m)
[servo_node_main-3]     | Source "/usr/include/c++/11/mutex", line 108, in __gthread_recursive_mutex_lock
[servo_node_main-3]     |   106:     lock()
[servo_node_main-3]     |   107:     {
[servo_node_main-3]     | > 108:       int __e = __gthread_recursive_mutex_lock(&_M_mutex);
[servo_node_main-3]     |   109: 
[servo_node_main-3]     |   110:       // EINVAL, EAGAIN, EBUSY, EINVAL, EDEADLK(may)
[servo_node_main-3]     | Source "/usr/include/x86_64-linux-gnu/c++/11/bits/gthr-default.h", line 811, in __gthread_mutex_lock
[servo_node_main-3]     |   809: __gthread_recursive_mutex_lock (__gthread_recursive_mutex_t *__mutex)
[servo_node_main-3]     |   810: {
[servo_node_main-3]     | > 811:   return __gthread_mutex_lock (__mutex);
[servo_node_main-3]     |   812: }
[servo_node_main-3]       Source "/usr/include/x86_64-linux-gnu/c++/11/bits/gthr-default.h", line 749, in stop [0x7f7bf4d7d7d2]
[servo_node_main-3]         746: __gthread_mutex_lock (__gthread_mutex_t *__mutex)
[servo_node_main-3]         747: {
[servo_node_main-3]         748:   if (__gthread_active_p ())
[servo_node_main-3]       > 749:     return __gthrw_(pthread_mutex_lock) (__mutex);
[servo_node_main-3]         750:   else
[servo_node_main-3]         751:     return 0;
[servo_node_main-3]         752: }
[servo_node_main-3] #5    Source "./nptl/pthread_mutex_lock.c", line 438, in __pthread_mutex_lock_full [0x7f7bf4518f50]
[servo_node_main-3] #4    Source "./assert/assert.c", line 101, in __assert_fail [0x7f7bf44bae95]
[servo_node_main-3] #3    Source "./assert/assert.c", line 92, in __assert_fail_base [0x7f7bf44a971a]
[servo_node_main-3] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f7bf44a97f2]
[servo_node_main-3] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f7bf44c3475]
[servo_node_main-3] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[servo_node_main-3]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[servo_node_main-3]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f7bf4517a7c]
[servo_node_main-3] Aborted (Signal sent by tkill() 16674 1000)

ibrahiminfinite, 2023-04-27:
@robertokcanale 
What version/commit is this on ?

ibrahiminfinite, 2023-05-16:
@robertokcanale 
Was this issue resolved ?

closed 2023-08-18 by github-actions[bot]

JaisonJose241, 2023-08-20:
Can you share the config file?