moveit/moveit2 #2152: Humble super slow planning time

trentatsa, 2023-05-02:
### Description

I have recently installed MoveIT on both ROS2 Foxy and Humble. During a basic path plan, Humble is taking up to 1 second to calculate a basic path plan, where as Foxy takes an order of magnitude less (0.1 seconds or less). Is anyone else having this issue ? 

### Your environment
* ROS Distro: Humble/Foxy
* OS Version: Ubuntu 22.04
* Humble = Build from Source, Foxy = Binary Install
* Cyclone DDS

henningkayser, 2023-05-25:
What planner and IK solver are you using? MoveGroup or MoveItCpp?

Make sure that you are [building Humble in Release mode](https://moveit.ros.org/install-moveit2/source/#build-moveit), you can also use [colcon mixins](https://github.com/colcon/colcon-mixin-repository/blob/master/build-type.mixin) for that:

```
colcon build --mixin release
```

trentatsa, 2023-05-27:
@henningkayser Thanks for the response. I am running the basic tutorial that uses Rviz. I'll make sure I build using mixin and get back to you.

leander2189, 2023-06-05:
Although I haven't done any formal benchmarking, I have noticed a performance increase after updating to the latest binaries...

I am curious, were there any performance issues with humble release?

henningkayser, 2023-06-05:
I am not aware of Humble-specific issues with performance. @leander2189 with "latest binaries", are you talking about Rolling or Iron?

leander2189, 2023-06-05:
> I am not aware of Humble-specific issues with performance. @leander2189 with "latest binaries", are you talking about Rolling or Iron?

No, I was talking about Humble.

If there is no humble-specific issues with performance, probably the performance changes in my setup are because of something else. Thanks for the info!

henningkayser, 2023-06-08:
Some actual benchmarks would help. Maybe @sjahr could set you up on how to run them.

closed 2023-09-08 by github-actions[bot]