moveit/moveit2 #2155: Defining max acceleration isn't documented

andrey-pr, 2023-05-06:
### Description

I got this string in logs:
```
Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
```
I have to set acceleration limits, but I don't know how to do it. Any docs of URDF doesn't say how to define acceleration in URDF. Also, I found examples of `joint_limits.yaml`, but I didn't find any examples of what to do with this file in launch file except examples of adding it to rviz configs, but I need to limit acceleration in planner, but not in rviz.
Thank you

### My environment
* ROS Distro: Humble

andrey-pr, 2023-05-10:
I realized, that I just need
```python
joint_limit_config = {
    "joint_limit_config": load_yaml(
        "robol_access_control_node",
        "config/joint_limits.yaml",
    )
}
```
or just call special function in moveit config builder

closed 2023-05-10 by andrey-pr

RobinHeitz, 2023-05-23:
Hey, 
I'm having the same problem. In the moveit docs I can only find sentences that "limits just need to be adjusted" but not what to do with this file.

My question is: Where are you using the 'joint_limit_config' parameters? In a MoveGroup Node? Somewhere else?
Thanks.

RobinHeitz, 2023-05-23:
So you are giving it the MoveGroup-Node as a parameter?

Thats exactly what I've tried, but it made no difference in the trajectory, even if i scaled velocity down with factor 10-100.

RobinHeitz, 2023-05-23:
Thats what I meant.
joint_limits.yaml is in /moveit_config/config.

load the yaml file and pass the content to the parameters[...] -tag of the move_group Node?

RobinHeitz, 2023-05-24:
Well thats not working for me.

Terminal yields that there are no acceleration parameters. See #2198

andrey-pr, 2023-05-25:
> Well thats not working for me.
> 
> Terminal yields that there are no acceleration parameters. See #2198

Answered in #2198

Karlosbubi, 2024-09-05:
> I realized, that I just need
> 
> ```python
> joint_limit_config = {
>     "robot_description_planning": load_yaml(
>         "robol_access_control_node",
>         "config/joint_limits.yaml",
>     )
> }
> ```
> 
> or just call special function in moveit config builder

Quick follow up question : What kind of ``load_yaml()` is that ? the text ? a dictionary of the values ? something ros specific ?
Thx