moveit/moveit2 #2156: MoveIt servo does not work with velocity controller

Pomorondza, 2023-05-07:
### Description

I cannot control the robot using MoveIt Servo with velocity_controllers/JointGroupVelocityController. Is it even supported by the newest version of MoveIt (Rolling/Humble)?

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* MoveIt source build main branch
* Robot: UR3e

### Expected behaviour
The robot follows the velocity commanded at the input (cartesian_command_in_topic).

### Actual behaviour
The robot starts moving in crazy configurations.

AndyZe, 2023-05-07:
Yes, it should be supported in Rolling/Humble

ibrahiminfinite, 2023-05-10:
@Pomorondza was your issue resolved?

Pomorondza, 2023-05-10:
Hi @ibrahiminfinite, yes and no, I will update my issue now if you could take a look again.

ibrahiminfinite, 2023-05-10:
> Hi @ibrahiminfinite, yes and no, I will update my issue now if you could take a look again.

I can take a look.
Which version of MoveIt are you using right now ?
Also is the `publish_joint_velocities` parameter set to `true` in your config file ?
Would be helpful if you can share your config file.

Pomorondza, 2023-05-10:
I am indeed using the source build of the main branch. Yes, I set the `publish_joint_velocities` to `true`. Here is my config file:
```#########################################
## MoveCore MoveItServo Parameters ##
#########################################
use_gazebo: false # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.01  # Max linear velocity. Meters per publish_period. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.5 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Rads or Meters per publish period. Only used for joint commands on joint_command_in_topic.
  joint: 0.1

# Optionally override Servo's internal velocity scaling when near singularity or collision (0.0 = use internal velocity scaling)
# override_velocity_scaling_factor = 0.0 # valid range [0.0:1.0]

## Properties of outgoing commands
publish_period: 0.02 # 1/Nominal publish rate [seconds]
low_latency_mode: false  # Set this to true to publish as soon as an incoming Twist command is received (publish_period is ignored)

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray or trajectory_msgs/JointTrajectory
command_out_type: std_msgs/Float64MultiArray

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: false
publish_joint_velocities: true
publish_joint_accelerations: false

## Plugins for smoothing outgoing commands
smoothing_filter_plugin_name: "online_signal_smoothing::ButterworthFilterPlugin"

# If is_primary_planning_scene_monitor is set to true, the Servo server's PlanningScene advertises the /get_planning_scene service,
# which other nodes can use as a source for information about the planning environment.
# NOTE: If a different node in your system is responsible for the "primary" planning scene instance (e.g. the MoveGroup node),
# then is_primary_planning_scene_monitor needs to be set to false.
is_primary_planning_scene_monitor: false

## Incoming Joint State properties
#low_pass_filter_coeff: 2.0  # Larger --> trust the filtered data more, trust the measurements less.

## MoveIt properties
move_group_name: ur_manipulator  # Often 'manipulator' or 'arm'
planning_frame: base_link  # The MoveIt planning frame. Often 'base_link' or 'world'

## Other frames
ee_frame_name: eef_tool  # The name of the tool link, used to return the EE pose
robot_link_command_frame: eef_tool  # commands must be given in the frame of a robot link. Usually either the base or tool

## Stopping behaviour
incoming_command_timeout:  5.0  # Stop servoing if X seconds elapse without a new command

## Configure handling of singularities and joint limits
lower_singularity_threshold:  100.0  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 200.0 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.
leaving_singularity_threshold_multiplier: 2.0 # Multiply the hard stop limit by this when leaving singularity (see https://github.com/ros-planning/moveit2/pull/620)

## Topic names
cartesian_command_in_topic: /servo/delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: /servo/delta_joint_cmds # Topic for incoming joint angle commands
joint_topic: /joint_states
status_topic: servo/status # Publish status to this topic
# Currently supported are /scaled_joint_trajectory_controller/joint_trajectory, /forward_position_controller/commands and /forward_velocity_controller/commands
command_out_topic: /forward_velocity_controller/commands # Publish outgoing commands here

## Collision checking for the entire robot body
check_collisions: false # Check collisions?
collision_check_rate: 10.0 # [Hz] Collision-checking can easily bog down a CPU if done too often.
self_collision_proximity_threshold: 0.01 # Start decelerating when a self-collision is this far [m]
scene_collision_proximity_threshold: 0.02 # Start decelerating when a scene collision is this far [m]
e_collision_distance: 0.01 # Stop if a collision is closer than this [m]```

ibrahiminfinite, 2023-05-10:
> For example, if the commanded linear speed at servo input is 10 mm/s, the linear speed at servo output (what is being sent to the controller) is around 4 mm/s

Asking since you mentioned 10 "mm/s"
The twist commands that are being sent to servo are expected to be in "m/s".  
A command scaling is applied here as well
https://github.com/ros-planning/moveit2/blob/237e9184c4c458f8e41f9eeb3f7cf3142e9e140e/moveit_ros/moveit_servo/src/servo_calcs.cpp#L908

Pomorondza, 2023-05-10:
Yes, yes, the commands I am sending are in m/s, the mm/s were just for reference.

I see... Maybe that is my problem, that the twist commands are not directly converted to joint velocities using the Jacobian, but first into delta Cartesian positions, then delta joint positions and then the velocity is somehow calculated from it. I will try to see what happens if I directly convert the twist command into joint velocities...

Pomorondza, 2023-05-10:
> * [OPEN] The robot adjusts to the target orientation only after the target position has been reached, and does so abruptly after some "thinking".
>   Does it have to do with the robot configurations?
Do you have any suggestions on this issue?

ibrahiminfinite, 2023-05-10:
> > * [OPEN] The robot adjusts to the target orientation only after the target position has been reached, and does so abruptly after some "thinking".
> >   Does it have to do with the robot configurations?
> 
> Do you have any suggestions on this issue?

This does not seem like an issue with servo itself.
Could possibly be a result of some optimization from the IK solver being used by the robot.

Pomorondza, 2023-05-11:
I managed to resolve what I wanted to accomplish (update in the first comment).

closed 2023-05-11 by Pomorondza

ibrahiminfinite, 2023-05-30:
@Pomorondza  your 3rd issue maybe related to this https://github.com/ros-planning/moveit2/pull/2211

Pomorondza, 2023-06-05:
> @Pomorondza your 3rd issue maybe related to this #2211

I believe my issue is just not "supported" at the moment in the sense that MoveIt Servo is meant to work with joint positions, and not joint velocities. I had to adapt this [function](https://github.com/ros-planning/moveit2/blob/dfd53ab9fc45ead1ce076be136564aeb83ab5643/moveit_ros/moveit_servo/src/utilities.cpp#L125) to take in joint velocities and send it directly to the controller.