moveit/moveit2 #2164: Non-chain robot with floating base

Acwok, 2023-05-10:
Hello,

My robot is a free-flyer cube with two ur5 arms. **I'd like to use one arm tip as a fixed base and the other as the end-effector**.
I've tried the main MoveIt 2 kinematics plugins (KDL, etc.) and none seems to support urdf with non-chain structure:
chain robot example: base -> ... -> tip_link
non-chain robot example: tip_link_1 -> ... -> base <- ... <- tip_link_2

**Do you have any solution to handle this kind of robot with two arms and a floating base ?**

Acwok, 2023-05-25:
**UPDATE:**
We started coding our own ik plugin based on [pinocchio](https://github.com/stack-of-tasks/pinocchio) which supports free-floating robots.  

We are trying to:
1) Send a goal position with respect to the world frame using `setPoseTarget()`
2) Our IK plugin computes a feasible configuration 
3) The planner fails to find a path 

The configuration is correct because we've tested it using `setJointValueTarget()` and the planner found a path.  
So we believe the planner can't find a path because it expects  the goal position to be with respect to the base_link and not to the world frame.  
This pb arises because we have a free floating base, when the base_link is the world frame there is no pb.

**Is it possible to use `setPoseTarget()` to send a goal pose with respect to another frame than the base_link ?**

closed 2023-08-25 by github-actions[bot]

7700770077, 2025-07-02:
Hello,

I encountered an issue while trying to use the Pinocchio IK plugin with MoveIt2. I have followed the setup process but am still unable to get it working.

Here is the kinematics.yaml file I generated using moveit_setup_assistant:

arm:
  kinematics_solver: kinematics_interface_pinocchio/KinematicsInterfacePinocchio
  kinematics_solver_search_resolution: 0.005
  kinematics_solver_timeout: 0.005

Despite installing the package with sudo apt-get install ros-humble-pinocchio, I am still facing issues whenever I attempt to use the Pinocchio solver.

Could you kindly guide me on how to properly integrate and utilize the Pinocchio plugin in MoveIt2?

Any help would be greatly appreciated!

Thank you.