moveit/moveit2 #2165: One of joints turns too quickly in servo mode

andrey-pr, 2023-05-10:
### Description

[image]
When I move end effector of my arm horizontally, one of joints (last in chain, joint which connects arm and gripper) turns too quickly, moves to end position and blocks arm. This joint have different from other joints maximum speed. In MoveGroup mode it works this problem not reproduces. What can I do to fix it? Thank you.

### My environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary build?
* Moveit 2.5.4
* Cyclone DDS

### Steps to reproduce
It can be reproduced only on my real hardware, not virtually.

### Expected behaviour
End effector must keep orientation, but it turns left when I move hand right

andrey-pr, 2023-05-11:
@AndyZe do you have any ideas?

AndyZe, 2023-05-11:
TBH it sounds like expected behavior. Servo is kind of dumb in that it only looks one step ahead; it is not able to predict something like this:

`moves to end position and blocks arm`

andrey-pr, 2023-05-11:
Problem not in `moves to end position and blocks arm`. Problem is that end effector must keep orientation, in this case it should be centered on image, but one of joints turned left and end effector turned left. I'll make video of in tomorrow and show you
 (i cannot reproduce it now, I need real robot to do it, and it stands in office)

AndyZe, 2023-05-11:
is it a 6-dof arm?

andrey-pr, 2023-05-11:
yes

AndyZe, 2023-05-11:
OK, that's good

andrey-pr, 2023-05-11:
I don't see any warnings. I'll make video of issue in rviz tomorrow

andrey-pr, 2023-05-12:
Hallo. I made video of it. https://file.io/fWXQss3YKWCu here is videos with rviz and with real robot. I sent on arm command to move linearly, so on video with rviz hand must move horizontally on screen without spinning, but it spins backward

andrey-pr, 2023-05-16:
@AndyZe have you any advice what to do with it?

ibrahiminfinite, 2023-05-18:
@andrey-pr 
> When I move end effector of my arm horizontally, one of joints (last in chain, joint which connects arm and gripper) turns too quickly, moves to end position and blocks arm.

What type of commands are you sending to Servo ? (JointJog / Twist )?

> Problem is that end effector must keep orientation, in this case it should be centered on image, but one of joints turned left and end effector turned left.

If you are sending JointJog or Twist, the robot simply takes single steps at each iteration, trying to reach the expected joint position. 
As far as I understand , this computation does not try to maintain an end effector pose.

If moving while maintaining end-effector pose is what you need, perhaps servoing using the  `PoseTracker` is the way to go for you
What do you think @AndyZe ?

AndyZe, 2023-05-18:
If you send a Twist command with zero rotational velocities, the orientation shouldn't change. That would be a pretty bad bug

Can I watch the video again @andrey-pr ? Looks like the file got deleted

andrey-pr, 2023-05-19:
Thank you. I realised that this is hardware problem, but not a problem of moveit. This problem exists when some of motors have too low velocity limit, and forward controller of servo does not check feedback and does not corrects other joints. I fixed my hardware, unlocked bigger velocities for all joints and It works

closed 2023-05-19 by andrey-pr