moveit/moveit2 #2178: Allow dynamically toggling collision checking in MoveIt Servo without stopping and restarting.

sea-bass, 2023-05-15:
I have bandwidth to work on this feature next week, but I am simply capturing this in an issue to open up discussions.

Now that we use `generate_parameter_library` and can easily change parameter values at runtime, this lets us be able to react to the `check_collisions` parameter more easily.

However, right now, the collision checking timer thread is only accessible in the start/stop service callbacks of MoveIt Servo. Meaning you can't dynamically enable/disable collision checking capabilities. A more performant runtime check could also be doable, so I'll be working on that.

ibrahiminfinite, 2023-05-16:
I was looking into this same thing.
It might also be good to not rely on the pub-sub mechanism and use a shared variable to update the `collision_velocity_scale` value instead.

mechwiz, 2023-05-16:
linking this issue as well https://github.com/ros-planning/moveit2/issues/409

sea-bass, 2023-05-21:
> I was looking into this same thing. It might also be good to not rely on the pub-sub mechanism and use a shared variable to update the `collision_velocity_scale` value instead.

I think this is a good idea performance-wise -- how about you make a separate issue for this if you'd like to tackle it as part of your refactoring efforts? Not super high priority, though.

AndyZe, 2023-05-21:
The original idea was that people would launch the servo node as a node component so pub/sub would be more efficient. There's a comment about that here:

https://github.com/ros-planning/moveit2/blob/243b0b2ccac156a75bcd158eefa86c8acf409110/moveit_ros/moveit_servo/launch/servo_example.launch.py#L80

I can't remember why L80 is commented, though. We should look into making a node component the default way. (Yet another issue...)

Agree, low priority.

closed 2023-05-22 by tylerjw