moveit/moveit2 #2182: Should 'moveit_setup_assistant' exec_depend on 'moveit_configs_utils'?

gavanderhoorn, 2023-05-17:
### Description

`apt install`-ing just `ros-humble-moveit-setup-assistant` and trying to `ros2 launch moveit_setup_assistant setup_assistant.launch.py` results in:

```
[INFO] [launch]: All log files can be found below /.../.ros/log/2023-05-17-11-37-59-890354-ecb941c48331-4395
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: No module named 'moveit_configs_utils'
```

I tried checking whether `moveit_configs_utils` is an `exec_depend` of one of the packages under `moveit_setup_assistant`.

On `master` it is, but only as a `test_depend` (`moveit_setup_controllers`).

On `humble` it doesn't seem to be mentioned at all in any manifest.

Is an `exec_depend` missing?

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* binaries, `apt`, `2.5.4`
* Which RMW: irrelevant, as this is the MSA not starting

gavanderhoorn, 2023-05-17:
seems it is a `depend` of `moveit_setup_app_plugins` on `main`, so perhaps this is just a `humble` issue?

gavanderhoorn, 2023-05-17:
Then again, `moveit_setup_assistant` directly depends on `moveit_configs_utils` (`main` and `humble`):

https://github.com/ros-planning/moveit2/blob/cc1ef618d9c02728bfc58723f15b29ed1efc6aa2/moveit_setup_assistant/moveit_setup_assistant/launch/setup_assistant.launch.py#L2-L5

so it should state its own dependencies.

henningkayser, 2023-05-25:
This looks like an `exec_depend` to me. Then, however, I'm not convinced that `add_debuggable_node` and `DeclareBooleanLaunchArg` are worth the dependency, if no config features are actually being used.

So.. https://github.com/ros-planning/moveit2/pull/1962 has been merged into main, but wasn't backported. I'll fix that.

PR #2206 (2023-05-25): add missing dependencies on config utils (backport #1962)
This is an automatic backport of pull request #1962 done by [Mergify](https://mergify.com).

Fixes #2182 
---

<details>
<summary>Mergify commands and options</summary>

<br />

More conditions and actions can be found in the [documentation](https://docs.mergify.com/).

You can also trigger Mergify actions by commenting on this pull request:

- `@Mergifyio refresh` will re-evaluate the rules
- `@Mergifyio rebase` will rebase this PR on its base branch
- `@Mergifyio update` will merge the base branch into this PR
- `@Mergifyio backport < [... cut]
closed 2023-08-25 by github-actions[bot]