moveit/moveit2 #2191: MoveIt Moving at vastly different speed than instructed

maxdgaa, 2023-05-20:
### Description

I'm using MoveIt servo to "hover over" a moving object which is moving in a straight line. The position of this moving object is known at all times. 

However, what I am seeing is that when I send a TwistStamped command so the arm will move in a straight line, the actual speed of the arm is vastly different than instructed. I've calculated it to be a ~2.3x difference. 

In another [post](https://answers.ros.org/question/411225/moveit-servo-does-not-follow-velocity-commmand-with-ur10e/) i read this might be due to the JointTrajectoryController not offering a _velocity_ state and command interface, but even when I add this:
```
# This config file is used by ros2_control
controller_manager:
  ros__parameters:
    update_rate: 100  # Control rate, match to robot.

    fanuc_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

fanuc_controller:
  ros__parameters:
    command_interfaces:
      - position
      - velocity
    state_interfaces:
      - position
      - velocity
    joints:
      - joint_1
      - joint_2
      - joint_3
      - joint_4
      - joint_5
      - joint_6
      ```
the problem still persists. I've already verified that the correct velocity commands are being sent but they are not being followed. Does anyone know what might be causing this?

This is my configuration file:
```
###############################################
# Modify all parameters related to servoing here
###############################################
use_gazebo: false # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s

# Optionally override Servo's internal velocity scaling when near singularity or collision (0.0 = use internal velocity scaling)
# override_velocity_scaling_factor = 0.0 # valid range [0.0:1.0]

## Properties of outgoing commands
publish_period: 0.034  # 1/Nominal publish rate [seconds]
low_latency_mode: false  # Set this to true to publish as soon as an incoming Twist command is received (publish_period is ignored)

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray or trajectory_msgs/JointTrajectory
command_out_type: trajectory_msgs/JointTrajectory

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: true
publish_joint_accelerations: false

## Plugins for smoothing outgoing commands
smoothing_filter_plugin_name: "online_signal_smoothing::ButterworthFilterPlugin"

# If is_primary_planning_scene_monitor is set to true, the Servo server's PlanningScene advertises the /get_planning_scene service,
# which other nodes can use as a source for information about the planning environment.
# NOTE: If a different node in your system is responsible for the "primary" planning scene instance (e.g. the MoveGroup node),
# then is_primary_planning_scene_monitor needs to be set to false.
is_primary_planning_scene_monitor: false

## MoveIt properties
move_group_name:  manipulator  # Often 'manipulator' or 'arm'
planning_frame: base_link  # The MoveIt planning frame. Often 'base_link' or 'world'
## Other frames
ee_frame_name: tool0  # The name of the end effector link, used to return the EE pose
robot_link_command_frame: tool0    # commands must be given in the frame of a robot link. Usually either the base or end effector

## Stopping behaviour
incoming_command_timeout:  0.10000 # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 4

## Configure handling of singularities and joint limits
lower_singularity_threshold:  17.0  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 30.0 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.
leaving_singularity_threshold_multiplier: 2.0 # Multiply the hard stop limit by this when leaving singularity (see https://github.com/ros-planning/moveit2/pull/620)

## Topic names
cartesian_command_in_topic: ~/delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: ~/delta_joint_cmds # Topic for incoming joint angle commands
joint_topic: /joint_states
status_topic: ~/status # Publish status to this topic
command_out_topic: /fanuc_controller/joint_trajectory # Publish outgoing commands here

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 10.0 # [Hz] Collision-checking can easily bog down a CPU if done too often.
self_collision_proximity_threshold: 0.01 # Start decelerating when a self-collision is this far [m]
scene_collision_proximity_threshold: 0.02 # Start decelerating when a scene collision is this far [m]
```

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* RMW: Cyclone

### Steps to reproduce
A geometry_msgs::msg::TwistStamped message is being sent to MoveIt servo. More specifically:
```
        auto msg = std::make_unique<geometry_msgs::msg::TwistStamped>();
        msg->header.stamp = MoveItInterfaceManager::getMoveItInterfaceManager().getROSNode()->now();
        msg->header.frame_id = "base_link";
        msg->twist.linear.x = 0; //m.getDisplacementInOneSecond().x() * multiplier * 3;
        msg->twist.linear.y = 0.01; //m.getDisplacementInOneSecond().y() * multiplier * 3;
        twist_cmd_pub_->publish(std::move(msg));
```

### Expected behaviour
The robot moves exactly at the instructed speed. 

### Actual behaviour
The robot moves significantly slower.

AndyZe, 2023-05-23:
I think the likely cause is applying the low-pass filtering in the position domain. We probably should filter in the velocity domain.

maxdgaa, 2023-05-27:
Hi @AndyZe, would you mind explaining a little bit on what that low pass filtering is in the servo context?

AndyZe, 2023-05-27:
The low-pass filter prevents sudden jerks in the input commands from being passed on to the robot.

@ibrahiminfinite is working on improving this right now but it will probably be a few months until it gets merged.

The specific line I'm talking about is here:  https://github.com/ros-planning/moveit2/blob/2ea784fd5d35637b22ad8bb3ae75273e4acd3a86/moveit_ros/moveit_servo/src/utilities.cpp#L143

The problem with filtering in the position domain is that, for constant robot velocity, the position is always changing. Thus the filter is always "damping" the motion. It would be better to filter in the velocity domain b/c the velocity would be constant.

If you want to make a PR to attempt this change, I'd be happy to review it. Otherwise, like I said, just wait a couple months and @ibrahiminfinite is likely to take care of it  :+1:

ibrahiminfinite, 2023-05-30:
The source of the problem was identified.
This PR https://github.com/ros-planning/moveit2/pull/2211 fixes it.
@osu-youa  @maxdgaa 
This should fix your issue once the PR is merged

maxdgaa, 2023-06-19:
Hi @ibrahiminfinite, 
Thanks for the reply. I looked over the PR but I don't think this fixes the issue since i'm using _speed_units_ instead of _unitless_. 

Kind regards

g-argyropoulos, 2023-07-21:
Hello @ibrahiminfinite,
Is there any hint or sample code that we could try to fix the issue ?

ibrahiminfinite, 2023-07-21:
@g-argyropoulos ,  if you are comfortable with testing out our development version.
It will be really helpful if you can try  the PR branch from #2224, it is an improved implementation.
And let us know if you still face the issue.

Thanks

closed 2023-12-01 by github-actions[bot]

BrocoliAsesino, 2025-01-17:
Hey @osu-youa ! Were you able to solve the scaling issue? I am facing a similar behavior in a UR5e. Even when I set very high proportional values in the PID of PoseTracking, the robot is unable to reach the target pose before a new one is published. New target pose publishing rate is low (30 Hz). I am not sure if the Servo is capping the joint velocities or is an issue of the controller. I am currently using position_controller/JointTrajectoryController (removes greatly the jittering as opposed to JointGroupVelocityController). Any hints on how to proceed would be great. Thanks in advance!

lukestroh, 2025-01-27:
@BrocoliAsesino -- @osu-youa is no longer working in the lab, but we still encounter the same issue using a UR5e using the forward_position_controller. Interestingly, when I test the robot with the launch flag `use_mock_hardware:=true`, the reduced speed bug does not occur.

AndyZe, 2025-01-27:
Quick guess - this is due to internal filtering of the UR5e. That filtering is external of MoveIt Servo, which is why it seems fine with `use_mock_hardware` (which bypasses the actual hardware).

lukestroh, 2025-01-27:
@AndyZe I would agree. Unfortunately the external scaling parameter (as set on the UR5e control pendant) is set to 100% scaling and I have yet to find any ros2-related UR source files that would correct this differently. Perhaps a low-level filter that we do not have access to