moveit/moveit2 #2192: OctoMap and Filtered_Cloud Not Updating During Movement Execution. Callback issue?

angeelalg, 2023-05-20:
### Description

I'm working with a 6-DOF robotic arm using MoveIt 2 in ROS 2 Foxy. The arm is equipped with a Stereolabs ZED2i camera, which I use for perception and to which I have previously applied hand-eye calibration.

I've followed the steps from the MoveIt Perception Pipeline tutorial to generate OctoMap from the ZED2i point cloud (which I've previously filtered to exclude parts of the robot's last link appearing in the OctoMap and simplified for faster operation). It works well, I can visualize the generation of the OctoMap in RVIZ2 and everything seems correct. If I send the robot a target that implies colliding with the environment, it rejects it.

The problem arises with dynamic obstacles. The OctoMap does not update while the robot is in motion. When a movement is executed, the entire OctoMap remains static without updating and it doesn't update again until the movement is completed, making it impossible to dodge obstacles and resulting in collisions.

Thanks to some answers provided by a user on ROS Answers, it was suggested that the problem might be related to not providing enough spinners for the callbacks. From what I've seen, the move_group node is implemented as multithreaded (using rclcpp::executors::MultiThreadedExecutor executor;), so I'm not entirely sure what could be going wrong. The thing is, if I launch two identical move_group nodes in the following way:

```
octomap_config = {'octomap_frame': 'zed2i_left_camera_frame', 
                  'octomap_resolution': 0.05,
                  'max_range': 2.0}
run_move_group_node_1 = Node(
        package="moveit_ros_move_group",
        executable="move_group",
        output="screen",
        parameters=[
            robot_description,
            robot_description_semantic,
            robot_description_kinematics,
            robot_description_planning,
            octomap_config,
            octomap_updater_config,
            ompl_planning_pipeline_config,
            trajectory_execution,
            moveit_controllers,
            planning_scene_monitor_parameters,
        ],
        arguments=["__log_level:=debug"],
    )
run_move_group_node_2 = Node(
        package="moveit_ros_move_group",
        executable="move_group",
        output="screen",
        parameters=[
            robot_description,
            robot_description_semantic,
            robot_description_kinematics,
            robot_description_planning,
            ompl_planning_pipeline_config,
            trajectory_execution,
            moveit_controllers,
            planning_scene_monitor_parameters,
        ],
        arguments=["__log_level:=debug"],
    )
```
The OctoMap is capable of updating during the movement of the arm (extremely slow, but it updates).
I believe it could be a problem related to Callback Groups or something similar; my knowledge of ROS2 is limited. I'm not sure if this is a bug or if it's a common occurrence in MoveIt.

I have also been able to verify that the filtered_cloud generated by MoveIt in its perception pipeline does not publish anything at all during the execution of movements.

```
sensors: ["zed2_handlerbot"]
zed2_handlerbot:
    sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
    point_cloud_topic: /zed2i_filtered_pointcloud
    max_range: 2.0
    point_subsample: 4
    padding_offset: 0.0
    padding_scale: 1.0
    max_update_rate: 2.0
    filtered_cloud_topic: filtered_cloud
```

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04.6 LTS
* Binary build
* Cyclone DDS

### Steps to reproduce
Follow the steps in the Perception Pipeline tutorial and try to update your OctoMap while making movements. 

### Expected behaviour
Try to avoid dynamic obstacles (whether by stopping or replanning, either way is fine with me).

### Actual behaviour
OctoMap does not update during the execution of arm movements, making it impossible to avoid collisions with dynamic obstacles.

I appreciate any insights or suggestions that anyone might have to help me resolve this issue. Thank you in advance for your time and support.

struggler-lu, 2023-06-11:
I also have the same problem. Have you solved it？

angeelalg, 2023-06-11:
Unfortunately, I haven't been able to solve the problem. I have tried many things, but nothing works.
The alternative I have ultimately adopted is to use the octomap from the MoveIt pipeline for planning, avoiding static obstacles, and additionally implementing another approach for dynamic obstacles:
Filtering the point cloud from my camera (keeping only the points within a distance of 1m) and performing clustering.
Generating collision objects at the positions of the clusters.
Adding the collision objects to the planning scene.

This way, dynamic collision objects can be visualized, and motion is stopped in case of a future collision, but there is no replanning.

struggler-lu, 2023-06-11:
Hi,Are you main that adding a collision objects like box at the positions of the clusters or directly adding point cloud to the collision objects? 
I also have an issue.After my obstacle moves, the point cloud will update, but the previously generated Octomap at position before obstacle movement   will still remain and cannot be automatically cleared.I use "rosservice info /clear_octomap" to solve this problem.
reference link：https://answers.ros.org/question/359503/what-is-the-intended-way-of-clearing-moveits-octomap/
Do you have any other solutions?

angeelalg, 2023-06-11:
Every 100ms, I remove all the collision objects present in my scene, then I take the clusters from the point cloud at that moment and generate the new collision objects. As for the octomap, the only way I know (at least with the octomap from the perception pipeline) is by calling the service you mentioned. Tomorrow, when I'm in the office, I can provide you with more information on how I do it all.

struggler-lu, 2023-06-12:
Thanks for your answer！

Zyonia, 2023-07-07:
> Every 100ms, I remove all the collision objects present in my scene, then I take the clusters from the point cloud at that moment and generate the new collision objects as 10cm boxes. As for the octomap, the only way I know (at least with the octomap from the perception pipeline) is by calling the service you mentioned. Tomorrow, when I'm in the office, I can provide you with more information on how I do it all.

Hi, I have the same problem, what's your progress now? Looking forward to your reply!