moveit/moveit2 #2197: Report code coverage correctly to SonarCloud

tylerjw, 2023-05-22:
We are currently running SonarCloud linting when there are pushes to this repo. We tried setting up reporting the code coverage details to SonarCloud but could not get it working correctly.

Tagging @EzraBrooks as he worked on this with me.

EzraBrooks, 2023-05-22:
I think the correct path here is to follow [this example using `gcovr`](https://github.com/sonarsource-cfamily-examples/linux-cmake-gcovr-gh-actions-sc), but we did not have time to continue investigating.

EzraBrooks, 2023-07-07:
@github-actions rude

@tylerjw I can point you at the stuff we did on Studio to make this work, it was a lot of trial and error to figure out exactly what was missing to cover ament_cmake C++, ament_cmake Python, and ament_python, but it's actually not much code.