moveit/moveit2 #2205: Speed-up clang-tidy + test config changes

henningkayser, 2023-05-25:
There are two somewhat major complaints around how we do clang-tidy testing:

1. We are missing clang-tidy fixes when changing the config, since we are only testing diffs
2. We are only testing diffs because full checks are horribly slow ([2h32m on a recent run](https://github.com/ros-planning/moveit2/actions/runs/5060355871/jobs/9083114050))

There supposedly is a low-hanging fruit that improves the situation, this [industrial_ci PR](https://github.com/ros-industrial/industrial_ci/pull/798) has been around for a while and speeds up runtime considerably (tested internally).
@EzraBrooks has also suggested just using [run_clang_tidy.py](https://clang.llvm.org/extra/doxygen/run-clang-tidy_8py_source.html) which would be just as easy, probably even reducing complexity of the CI config.

130s, 2023-05-25:
> There supposedly is a low-hanging fruit that improves the situation, this https://github.com/ros-industrial/industrial_ci/pull/798 has been around for a while and speeds up runtime considerably (tested internally).

Oops, that's a shame (as one of the original authors). I'll look.

EzraBrooks, 2023-07-10:
this is a really aggressive stale bot