moveit/moveit2 #2209: Planning with a floating base and a fixed end-effector

Acwok, 2023-05-30:
**Problem**
My robot is a free-floating cube with two ur5 arms. I'd like to use one arm tip as a fixed base and the other as the end-effector but I can't because the planner always consider the **base_link** (the cube) as fixed.

For example, in this video the robot can't avoid obstacles because the planner seems to plan with the base (cube) being fixed to its position (see the "ghost" of the planned trajectory):

https://github.com/ros-planning/moveit2/assets/7405791/c9baf0ce-491b-46b5-be83-1e57fb08f3db

**Solution**
We would like a function allowing to specify the link we want to fix during the planning.  
In our case, it would be one end-effector (instead of the cube).

**Context**
We are using ROS Humble on ubuntu 22.04 with Moveit2.

henningkayser, 2023-06-05:
I think this won't work as long as the cube is considered the "base" (or root link) of the planning group that you use. If you want the tip of one arm to be "fixed", you should define that one as the root link and change the cube to be a simple robot link connected with fixed joints.

Acwok, 2023-06-05:
Currently my URDF looks like this:
```
           cube
        /        \
left_arm       right_arm
     |              |
left_wrist     right_wrist
```
Do you mean I have to change the URDF to a direct chain ?
```
left_wrist > left_arm > cube > right_arm > right_wrist
```

Or do you mean there is a way to set the **left_wrist** as root link in the SRDF when defining planning groups ?
If so, I'm really interested because I haven't found how.

Acwok, 2023-06-05:
I can provide the URDF and SRDF I'm using:
[my_robot.urdf](https://github.com/ros-planning/moveit2/files/11654940/my_robot.urdf.txt)
[moveit_marm.srdf](https://github.com/ros-planning/moveit2/files/11654948/moveit_marm.srdf.txt)

In my opinion, It would be interesting to have the possibility to set the root planning frame to use in MoveIt without having to modify the URDF. Maybe it is already possible ?  
If not (and if it is not too much work), maybe I can help developing the feature because I feel everything is almost there.

Acwok, 2023-06-21:
@henningkayser I'm not sure to understand your previous answer, could you please provide more details ?

closed 2023-09-21 by github-actions[bot]