moveit/moveit2 #2215: Setting use_sim_time to true does not generate octomap.

kvkpraneeth, 2023-05-31:
### Description

I am trying to setup MoveIt2 with a UR5e in Gazebo Fortress. Everything works well apart from the sensors. I have a depth camera that also publishes `sensor_msgs/PointCloud2`, Since I am running a simulation and the `/clock` topic is present, I passed `use_sim_time:=true` to the move group node. In my observation after passing this parameter the octomaps do not get published. If I pass `use_sim_time:=false` to the move group node the octomaps get published.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Binary
* Which RMW (Fast DDS or Cyclone DDS)? Cyclone

### Steps to reproduce
sensors_3d.yaml:
```yaml
sensors:
  - default_sensor
default_sensor:
  filtered_cloud_topic: filtered_cloud
  max_range: 8.0
  max_update_rate: 1.0
  padding_offset: 0.1
  padding_scale: 1.0
  point_cloud_topic: /depth_camera/points
  point_subsample: 1
  sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater

octomap_frame: camera
octomap_resolution: 0.05
max_range: 8.0
```

move_group.launch.py
```python3
from moveit_configs_utils import MoveItConfigsBuilder
from moveit_configs_utils.launches import *

def generate_move_group_launch(moveit_config):
    ld = LaunchDescription()

    ld.add_action(DeclareBooleanLaunchArg("debug", default_value=False))
    ld.add_action(
        DeclareBooleanLaunchArg("allow_trajectory_execution", default_value=True)
    )
    ld.add_action(
        DeclareBooleanLaunchArg("publish_monitored_planning_scene", default_value=True)
    )
    # load non-default MoveGroup capabilities (space separated)
    ld.add_action(DeclareLaunchArgument("capabilities", default_value=""))
    # inhibit these default MoveGroup capabilities (space separated)
    ld.add_action(DeclareLaunchArgument("disable_capabilities", default_value=""))

    # do not copy dynamics information from /joint_states to internal robot monitoring
    # default to false, because almost nothing in move_group relies on this information
    ld.add_action(DeclareBooleanLaunchArg("monitor_dynamics", default_value=False))

    should_publish = LaunchConfiguration("publish_monitored_planning_scene")

    move_group_configuration = {
        "publish_robot_description_semantic": True,
        "allow_trajectory_execution": LaunchConfiguration("allow_trajectory_execution"),
        # Note: Wrapping the following values is necessary so that the parameter value can be the empty string
        "capabilities": ParameterValue(
            LaunchConfiguration("capabilities"), value_type=str
        ),
        "disable_capabilities": ParameterValue(
            LaunchConfiguration("disable_capabilities"), value_type=str
        ),
        # Publish the planning scene of the physical robot so that rviz plugin can know actual robot
        "publish_planning_scene": should_publish,
        "publish_geometry_updates": should_publish,
        "publish_state_updates": should_publish,
        "publish_transforms_updates": should_publish,
        "monitor_dynamics": False,
        "use_sim_time": True
    }

    move_group_params = [
        moveit_config.to_dict(),
        move_group_configuration,
    ]

    add_debuggable_node(
        ld,
        package="moveit_ros_move_group",
        executable="move_group",
        commands_file=str(moveit_config.package_path / "launch" / "gdb_settings.gdb"),
        output="screen",
        parameters=move_group_params,
        extra_debug_args=["--debug"],
        # Set the display variable, in case OpenGL code is used internally
        additional_env={"DISPLAY": ":0"},
    )
    return ld

def generate_launch_description():
    moveit_config = MoveItConfigsBuilder("ur", package_name="ur_moveit").to_moveit_configs()
    return generate_move_group_launch(moveit_config)
```

### Expected behaviour
[image]

This octomap should be produced (Screenshot taken when `use_sim_time:=false`)

### Actual behaviour

[image]

There is no octomap being published.

dazednconfusing, 2023-10-05:
The issue for me was that ignition gazebo was not publishing to /clock. Adding this to the ros gz bridge config file:
```
- ros_topic_name: '/clock'
  ros_type_name: 'rosgraph_msgs/msg/Clock'
  gz_type_name: 'ignition.msgs.Clock'
  direction: GZ_TO_ROS
```
 worked for me. Or equivalently, in a separate terminal run:
`ros2 run ros_gz_bridge parameter_bridge /clock@rosgraph_msgs/msg/Clock[ignition.msgs.Clock`

kvkpraneeth, 2023-10-07:
That was sloppy of me! This fixed the issue, thanks for pointing it out @dazednconfusing. Closing this issue now.

closed 2023-10-07 by kvkpraneeth