moveit/moveit2 #2220: MoveItPy does not support use_sim_time:=True

henningkayser, 2023-06-01:
### Description

When working around #2219 by merging `use_sim_time:=True` into the MoveIt config dictionary, I'm seeing mixed results:

* `use_sim_time` is actually set to `True` :tada: 
* The moveit_py node is crashing due to an `rclcpp::exceptions::InvalidParameterValueException`

In particular it prints:
```
[motion_planning_python_api_tutorial.py-2] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
[motion_planning_python_api_tutorial.py-2]   what():  parameter 'qos_overrides./clock.subscription.durability' could not be set:
```

A quick search brought me to [this issue comment](https://github.com/ros2/rclcpp/issues/1566#issuecomment-788779572), pointing out that `use_sim_time:=True` will enable qos_override callbacks on the `/clock` topic. Question is, why does everything work with the C++ versions MoveGroup/MoveItCpp, and why do we run into this exception with moveit_py.
I have briefly skimmed the node initialization and didn't find any obvious issues.

For context, I've been working on a MoveIt+Gazebo setup and wanted to add a MoveItPy example.
[Here](https://github.com/henningkayser/moveit2_tutorials/blob/panda_gz/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.rst#running-the-tutorial-in-a-gazebo-simulation) are instructions for running different launch files in a docker compose container. You'll find that all service targets besides **python_gazebo** work, **python_gazebo** fails for the described reason.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* [jammy-humble](https://github.com/moveit/moveit2_packages/tree/jammy-humble) binaries (main branch)

### Steps to reproduce
Try passing `use_sim_time:=True` [like this](https://github.com/henningkayser/moveit2_tutorials/blob/panda_gz/doc/examples/motion_planning_python_api/launch/motion_planning_python_api_tutorial.launch.py#L67).

### Expected behaviour
The launch file should work and allow using sim time with moveit_py.

### Actual behaviour
The Exception is thrown, node dies.

sea-bass, 2023-06-02:
Do things actually work with `MoveItCpp`? Like if we launch the [MoveItCpp tutorial](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/examples/moveit_cpp/launch/moveit_cpp_tutorial.launch.py#L26) and similarly test with `use_sim_time` as in these 2 issues, are the errors the same or not?

And if they aren't, the only difference I see in [the tutorial](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/examples/moveit_cpp/src/moveit_cpp_tutorial.cpp#L21-L28) vs. [MoveItPy[(https://github.com/ros-planning/moveit2/blob/main/moveit_py/src/moveit/moveit_ros/moveit_cpp/moveit_cpp.cpp#L86-L88) is the addition of `allow_undeclared_parameters(true)` ... which I don't know if it will do much?

henningkayser, 2023-06-02:
@sea-bass thanks for testing.. Yes, MoveGroup works and MoveItCpp should work as well since that's being used internally. The `allow_undeclared_parameters(true)` flag is also used by MoveGroup [here](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/move_group/src/move_group.cpp#L213), so that shouldn't be the issue.
My current guess is that MoveItPy is [missing some ROS args](https://github.com/ros-planning/moveit2/blob/main/moveit_py/src/moveit/moveit_ros/moveit_cpp/moveit_cpp.cpp#L64) for the [global context](https://github.com/ros2/rclcpp/blob/33dae5d679751b603205008fcb31755986bcee1c/rclcpp/src/rclcpp/utilities.cpp#L41) (whatever that really means...)

henningkayser, 2023-06-02:
ok, looks like rclcpp is really not happy about being told about `use_sim_time:=true`. #2223 fixes moveit_py sim for me.

closed 2023-06-20 by JafarAbdi

jacob-kruse, 2024-11-13:
Now that this is fixed, how do I properly pass the use_sim_time=True to MoveItPy? @henningkayser

PR #2223 (2023-06-20): Fix moveit_py rclcpp::init()
Fixes #2220. Rclcpp has been initialized without args which was problematic for some use cases like clock simulation. Parameters like use_sim_time:=true need to be passed to rclcpp, also NodeOptions access the global rcl state on construction.

@peterdavidfagan maybe we should just pass all the launch arguments from moveit_py over to rclcpp through the MoveItPy constructor. That would also take care of #2219