moveit/moveit2 #2221: Moveit Python in preview versions

VegaDeg, 2023-06-02:
Hi, thanks for the great work in creating the python bindings for Moveit2. However, it seems that the binding is only available in the main branch, and I failed to build it under ROS2 humble, I wonder if it will be propagated into previous versions? Or is it possible to build the main branch in humble?

adamwhats, 2023-07-07:
@henningkayser I've been using the nightly debians without an issue the last few days, but am having some problems today. When I add the repos and run `apt update` I get `404  Not Found [IP: 185.199.109.133 443]`. Has anything changed in the last day or two?

Full output:
```
ros@a8f6e37ca815:/workspaces/moveit_py_test$ echo "deb [trusted=yes] https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble/ ./" \
    | sudo tee /etc/apt/sources.list.d/moveit-moveit2-packages.list
deb [trusted=yes] https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble/ ./
ros@a8f6e37ca815:/workspaces/moveit_py_test$ echo "yaml https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble/local.yaml humble" \
    | sudo tee /etc/ros/rosdep/sources.list.d/1-moveit-moveit2-packages.list
yaml https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble/local.yaml humble
ros@a8f6e37ca815:/workspaces/moveit_py_test$ sudo apt update
Hit:2 https://ppa.launchpadcontent.net/kisak/kisak-mesa/ubuntu jammy InRelease                                                                                                   
Hit:3 http://security.ubuntu.com/ubuntu jammy-security InRelease                                                                                                                 
Hit:4 http://archive.ubuntu.com/ubuntu jammy InRelease                                    
Ign:1 https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble ./ InRelease
Hit:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease      
Hit:7 http://packages.ros.org/ros2/ubuntu jammy InRelease           
Hit:8 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Ign:5 https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble ./ Release
Err:9 https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble ./ Packages
  404  Not Found [IP: 185.199.109.133 443]
Err:9 https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble ./ Packages
  404  Not Found [IP: 185.199.109.133 443]
Err:9 https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble ./ Packages
  404  Not Found [IP: 185.199.109.133 443]
Err:9 https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble ./ Packages
  404  Not Found [IP: 185.199.109.133 443]
Err:9 https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble ./ Packages
  404  Not Found [IP: 185.199.109.133 443]
Err:9 https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble ./ Packages
  404  Not Found [IP: 185.199.109.133 443]
Err:9 https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble ./ Packages
  404  Not Found [IP: 185.199.109.133 443]
Reading package lists... Done
E: Failed to fetch https://raw.githubusercontent.com/moveit/moveit2-packages/jammy-humble/Packages  404  Not Found [IP: 185.199.109.133 443]
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

adamwhats, 2023-07-08:
My bad, I've found the issue. I managed to swap `.../moveit2_packages/jammy-humble/...` to `.../moveit2-packages/jammy-humble/...` in the URLs :facepalm:

closed 2023-12-01 by github-actions[bot]