moveit/moveit2 #2225: Use rclcpp::SystemDefaultsQoS() instead of magic numbers (or document magic numbers)

sjahr, 2023-06-05:
In many places throughout the codebase, we're still using magic numbers to configure QoS for publishers and subscribers for example:
[here](https://github.com/ros-planning/moveit2/blob/55df0bccd5e884649780b4ceeee80891e563b57b/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor_middleware_handle.cpp#L77), [here](https://github.com/ros-planning/moveit2/blob/55df0bccd5e884649780b4ceeee80891e563b57b/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp#L1130) or [here](https://github.com/ros-planning/moveit2/blob/55df0bccd5e884649780b4ceeee80891e563b57b/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp#L186).

I think we should replace this in all cases with rclcpp::SystemDefaultsQoS() for consistency and maybe more stability. @henningkayser + @tylerjw What do you think?

SANTHOSH-MAMIDISETTI, 2023-06-16:
@henningkayser  @sjahr  I would love to work on it , Can I be assigned it ?

sjahr, 2023-06-30:
You can just start working on it :smile: Once you've opened a PR we can link it to this issue

Shobuj-Paul, 2023-07-23:
Hi @sjahr 
Could you check the draft PR if this was what you were looking for? If this is correct I shall look for more instances and make the changes.

PR #2271 (2023-07-31): Replaced numbers with SystemDefaultsQos()
### Description

This PR replaces "magic numbers" with SystemDefaultsQos(), making it more readable and possibly more stable.
Fixes #2225

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/ros-planning/moveit2/blob/main/MIGRATION.md) notes
- [ ] Create tests, which fail wit [... cut]
closed 2023-07-31 by henningkayser