moveit/moveit2 #2227: Reduce dependency on boost by replacing boost::algorithm::join

sjahr, 2023-06-06:
We're using `boost::algorithm::join(...)` in multiple places throughout the codebase to concatenate strings. It would be nice to remove this for example by replacing it with a small custom function based on the std library placed in moveit_core/utils

tylerjw, 2023-06-06:
Of all the things in Boost we depend on the algorithm is a nice library. We could write our own join but doing so at the quality of the one in Boost would be time consuming and would just increase the amount of code we have to maintain.

JafarAbdi, 2023-06-06:
How about using [fmt::join](https://fmt.dev/latest/api.html?highlight=join#_CPPv4I0EN3fmt4joinE9join_viewIN6detail10iterator_tI5RangeEEN6detail10sentinel_tI5RangeEEERR5Range11string_view) for this? It would be nice if we could slowly start using that library for string formatting

tylerjw, 2023-06-06:
I like the idea of moving towards using `fmt` over boost for string formatting. Especially since that is the library that got standardized and it is much easier to include and inform your users about it than boost.

Shobuj-Paul, 2023-07-26:
I tried replacing `boost::algorithm::join` with `fmt::format` and `fmt::join` after including `fmt/format.h` header, but there is the following error during build.
```bash
--- stderr: moveit_ros_occupancy_map_monitor
/usr/bin/ld: CMakeFiles/occupancy_map_monitor_tests.dir/test/occupancy_map_monitor_tests.cpp.o: in function `testing::internal::ActionResultHolder<occupancy_map_monitor::OccupancyMapMonitor::Parameters>::PrintAsActionResult(std::ostream*) const':
occupancy_map_monitor_tests.cpp:(.text._ZNK7testing8internal18ActionResultHolderIN21occupancy_map_monitor19OccupancyMapMonitor10ParametersEE19PrintAsActionResultEPSo[_ZNK7testing8internal18ActionResultHolderIN21occupancy_map_monitor19OccupancyMapMonitor10ParametersEE19PrintAsActionResultEPSo]+0x39): undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
/usr/bin/ld: CMakeFiles/occupancy_map_monitor_tests.dir/test/occupancy_map_monitor_tests.cpp.o: in function `testing::internal::FunctionMocker<void (std::function<bool (std::shared_ptr<rmw_request_id_s>, std::shared_ptr<moveit_msgs::srv::SaveMap_Request_<std::allocator<void> > >, std::shared_ptr<moveit_msgs::srv::SaveMap_Response_<std::allocator<void> > >)>)>::UntypedPrintArgs(void const*, std::ostream*) const':
occupancy_map_monitor_tests.cpp:(.text._ZNK7testing8internal14FunctionMockerIFvSt8functionIFbSt10shared_ptrI16rmw_request_id_sES3_IN11moveit_msgs3srv16SaveMap_Request_ISaIvEEEES3_INS7_17SaveMap_Response_IS9_EEEEEEE16UntypedPrintArgsEPKvPSo[_ZNK7testing8internal14FunctionMockerIFvSt8functionIFbSt10shared_ptrI16rmw_request_id_sES3_IN11moveit_msgs3srv16SaveMap_Request_ISaIvEEEES3_INS7_17SaveMap_Response_IS9_EEEEEEE16UntypedPrintArgsEPKvPSo]+0x31): undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
/usr/bin/ld: CMakeFiles/occupancy_map_monitor_tests.dir/test/occupancy_map_monitor_tests.cpp.o: in function `testing::internal::FunctionMocker<void (std::function<bool (std::shared_ptr<rmw_request_id_s>, std::shared_ptr<moveit_msgs::srv::LoadMap_Request_<std::allocator<void> > >, std::shared_ptr<moveit_msgs::srv::LoadMap_Response_<std::allocator<void> > >)>)>::UntypedPrintArgs(void const*, std::ostream*) const':
occupancy_map_monitor_tests.cpp:(.text._ZNK7testing8internal14FunctionMockerIFvSt8functionIFbSt10shared_ptrI16rmw_request_id_sES3_IN11moveit_msgs3srv16LoadMap_Request_ISaIvEEEES3_INS7_17LoadMap_Response_IS9_EEEEEEE16UntypedPrintArgsEPKvPSo[_ZNK7testing8internal14FunctionMockerIFvSt8functionIFbSt10shared_ptrI16rmw_request_id_sES3_IN11moveit_msgs3srv16LoadMap_Request_ISaIvEEEES3_INS7_17LoadMap_Response_IS9_EEEEEEE16UntypedPrintArgsEPKvPSo]+0x31): undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
/usr/bin/ld: CMakeFiles/occupancy_map_monitor_tests.dir/test/occupancy_map_monitor_tests.cpp.o: in function `testing::internal::ActionResultHolder<std::shared_ptr<occupancy_map_monitor::OccupancyMapUpdater> >::PrintAsActionResult(std::ostream*) const':
occupancy_map_monitor_tests.cpp:(.text._ZNK7testing8internal18ActionResultHolderISt10shared_ptrIN21occupancy_map_monitor19OccupancyMapUpdaterEEE19PrintAsActionResultEPSo[_ZNK7testing8internal18ActionResultHolderISt10shared_ptrIN21occupancy_map_monitor19OccupancyMapUpdaterEEE19PrintAsActionResultEPSo]+0x6c): undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
/usr/bin/ld: CMakeFiles/occupancy_map_monitor_tests.dir/test/occupancy_map_monitor_tests.cpp.o: in function `testing::internal::FunctionMocker<void (std::shared_ptr<occupancy_map_monitor::OccupancyMapUpdater>)>::UntypedPrintArgs(void const*, std::ostream*) const':
occupancy_map_monitor_tests.cpp:(.text._ZNK7testing8internal14FunctionMockerIFvSt10shared_ptrIN21occupancy_map_monitor19OccupancyMapUpdaterEEEE16UntypedPrintArgsEPKvPSo[_ZNK7testing8internal14FunctionMockerIFvSt10shared_ptrIN21occupancy_map_monitor19OccupancyMapUpdaterEEEE16UntypedPrintArgsEPKvPSo]+0x6e): undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
/usr/bin/ld: CMakeFiles/occupancy_map_monitor_tests.dir/test/occupancy_map_monitor_tests.cpp.o:occupancy_map_monitor_tests.cpp:(.text._ZNK7testing8internal14FunctionMockerIFvSt8functionIFbSt10shared_ptrI16rmw_request_id_sES3_IN11moveit_msgs3srv16LoadMap_Request_ISaIvEEEES3_INS7_17LoadMap_Response_IS9_EEEEEEE32UntypedDescribeUninterestingCallEPKvPSo[_ZNK7testing8internal14FunctionMockerIFvSt8functionIFbSt10shared_ptrI16rmw_request_id_sES3_IN11moveit_msgs3srv16LoadMap_Request_ISaIvEEEES3_INS7_17LoadMap_Response_IS9_EEEEEEE32UntypedDescribeUninterestingCallEPKvPSo]+0x7b): more undefined references to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)' follow
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/occupancy_map_monitor_tests.dir/build.make:344: occupancy_map_monitor_tests] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:232: CMakeFiles/occupancy_map_monitor_tests.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_ros_occupancy_map_monitor [0.94s, exited with code 2]
```

This looks like I need to include fmt as a dependency somehere. Can someone confirm?

sjahr, 2023-07-27:
Thanks, @Shobuj-Paul! Would you mind opening a draft PR? That way it is easier to see what you've changed so far and where something might be missing.

sjahr, 2023-07-27:
You need to properly link the targets in the CMakeLists.txt files against fmt like for example here: https://github.com/PickNikRobotics/pick_ik/blob/main/CMakeLists.txt. Also, please update the package.xml files

PR #2273 (2023-08-12): Replaced boost::algorithm::join with fmt::join
### Description

This PR replaces `boost::algorithm::join` with `fmt::join` for string formatting to reduce boost dependencies.
Fixes #2227 

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/ros-planning/moveit2/blob/main/MIGRATION.md) notes
- [ ] Create tests, which  [... cut]
closed 2023-08-12 by sjahr