moveit/moveit2 #2231: Circular and curve movements

andrey-pr, 2023-06-06:
Hallo. I can not find any docs about MoveGroup movements of circular or curve trajectory movements/ As I guess, there is no any solution. For example, I want to make circular trajectory movements to stir a cup of coffee. How can I do it? Currently, I'm trying to do it by passing hundreds of points to computeCartesianPath, but its not a good way and it I still have problems with it. @AndyZe, do you have any better solution?

andrey-pr, 2023-06-13:
> You could try to use the PILZ motion planner's [CIRC mode](https://moveit.picknik.ai/main/doc/how_to_guides/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html?highlight=pilz#the-circ-motion-command). PILZ is a Cartesian trajectory generator (It can generate trajectories in free space (essentially a Cartesian interpolation) but cannot do Cartesian motion planning (e.g. obstacle avoidance)). Checkout the tutorial code on how to use it: https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/how_to_guides/pilz_industrial_motion_planner/src/pilz_move_group.cpp

Thank you

sjahr, 2023-06-15:
I hope this solved your problem. Feel free to re-open this issues if you feel it is not resolved or open a new one for a new question

closed 2023-06-15 by sjahr