moveit/moveit2 #2250: Cannot import moveit_py

faheemkhaskheli9, 2023-06-26:
I create a ros2 package, ros2 node working fine. 
Trying to import moveit_py.

Import moveit
# error moveit not found

Import moveit_py
#error not found

Import MoveItPy
# error not found. 

I sourced moveit_ws, still not working. 
Created project in moveit ws, still not working.

faheemkhaskheli9, 2023-06-26:
Trying to run this code. 

```Python
rclpy.init()
logger = rclpy.logging.get_logger("moveit_py.pose_goal")

# instantiate MoveItPy instance and get planning component
panda = MoveItPy(node_name="moveit_py")
panda_arm = panda.get_planning_component("panda_arm")
logger.info("MoveItPy instance created")

```

closed 2023-06-26 by faheemkhaskheli9

RobinHeitz, 2023-10-13:
Can you tell me what you've found out? Am sitting at the same problem.