moveit/moveit2 #2254: Getting parsing error with moveit python api

ZealousGinger, 2023-06-28:
## Description

I get this error when trying to use the moveit python api following [this motion planning tutorial](https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html#getting-started)

```
RuntimeError: failed to initialize rcl: Couldn't parse params file: '--params-file Failed to parse params file path from command line arguments. Check that --params-file command line argument is specified.'. Error: Error opening YAML file, at ./src/parser.c:270, at ./src/rcl/arguments.c:406
[ros2run]: Process exited with failure 1
```
## Your environment

    ROS Distro: Humble
    OS Version: Ubuntu 22.04
    Moveit2: 
    Binary:Humble
    Source:Main(rolling)
    RMW:cyclone

I am unsure why i get this paring error, even when i followed the tutorial.
I am new to moveit2 and would greatly appreciate any help!

JafarAbdi, 2023-06-28:
I think it was introduced by https://github.com/ros-planning/moveit2/pull/2223, do you mind testing with that commit reverted? I'll take a look later today

ZealousGinger, 2023-06-28:
> I think it was introduced by #2223, do you mind testing with that commit reverted? I'll take a look later today

Hi Im pretty new to this so Im not sure how to revert a commit, may I know how do i do it ? Thank you !

ZealousGinger, 2023-06-30:
> I think it was introduced by #2223, do you mind testing with that commit reverted? I'll take a look later today

Hey, i tried with the reverted code, but its still giving the same error. I also tried using the launch_params_filepath parameter, and still gives the same error. May I know if there is advice for this, thank you !

ZealousGinger, 2023-07-03:
I managed to fix this problem by using the MoveItConfigsBuilder but now i get another error saying 
```
[ERROR] [1688372176.747794913] [moveit.ros_planning_interface.moveit_cpp]: Failed to load any planning pipelines.
[FATAL] [1688372176.747824181] [moveit.ros_planning_interface.moveit_cpp]: Failed to load planning pipelines from parameter server

```
similar to another issue in
#2014

closed 2023-07-03 by ZealousGinger

RemiFabre, 2023-10-06:
> I managed to fix this problem by using the MoveItConfigsBuilder but now i get another error saying
>
> ```
> [ERROR] [1688372176.747794913] [moveit.ros_planning_interface.moveit_cpp]: Failed to load any [...]

Did you fix this issue? I'm having the exact same message

ymollard, 2025-01-25:
```
[ERROR] [1688372176.747794913] [moveit.ros_planning_interface.moveit_cpp]: Failed to load any planning pipelines.
[FATAL] [1688372176.747824181] [moveit.ros_planning_interface.moveit_cpp]: Failed to load planning pipelines from parameter server
```
That seems related to a wrong `MoveItConfigsBuilder` passed to your node.
Check that you have passed the configuration of the planning pipelines to `moveit_cpp`:
https://github.com/moveit/moveit2_tutorials/blob/12edc7ea652a6590bda0fc0dc1cddfe8f0402748/doc/examples/motion_planning_python_api/launch/motion_planning_python_api_tutorial.launch.py#L22-L23

If you launch your code from a launchfile, that `MoveItConfigsBuilder` declaration should be passed as parameters from the launchfile as in the example:
https://github.com/moveit/moveit2_tutorials/blob/12edc7ea652a6590bda0fc0dc1cddfe8f0402748/doc/examples/motion_planning_python_api/launch/motion_planning_python_api_tutorial.launch.py

If you launch the node with `ros2 run`, that should be in the config dict:
`MoveItPy(node_name="moveit_py", config_dict=moveit_config)`

If you don't have such configuration yet, you can rely on the example from the tutorials that initializes default config for OMPL, CHOMP and PLP : https://github.com/moveit/moveit2_tutorials/blob/12edc7ea652a6590bda0fc0dc1cddfe8f0402748/doc/examples/motion_planning_python_api/config/motion_planning_python_api_tutorial.yaml