moveit/moveit2 #2256: Have MoveIt Setup Assistant generate ompl_planning.yaml file

reidchristopher, 2023-06-28:
**Is your feature request related to a problem? Please describe.**
The ROS 1 iteration of MoveIt Setup Assistant generated an ompl_planning.yaml file, while the current ROS 2 version does not. This ompl_planning.yaml is still used for configuration in the current state of MoveIt, so producing this file would make the setup assistant that much more user friendly.

**Describe the solution you'd like**
It would be great for the setup assistant to auto-generate some default version of  ompl_planning.yaml. Essentially combining the defaults in `moveit_config_utils` and adding on the small amount of necessary information (listing all of the available configs for each planning group, projection_evaluator, default_planner_config, etc). Most of the pieces are already in place, it's just putting it together in a way that isn't broken.

**Describe alternatives you've considered**
You can load the defaults with `MoveItConfigBuilder` and then modify the configuration, but this already assumes more knowledge of the expected format, and is just unnecessary extra work on the user's part. You can also copy from the default files in `moveit_config_utils`, but that again is a clunky way of solving the issue.

**Additional context**
Would also make the "Group Default Planner" button do something when defining planning groups, since I'm pretty sure that doesn't do anything at the moment. :eyes:

sjahr, 2023-07-12:
@reidchristopher Sounds like a good idea to add this :+1: Would you have time to work on this? I can assist you with reviews and answering questions

obensustam2, 2023-07-21:
@reidchristopher It is a very good idea 👍🏼

I tried to implement ompl_planning.yaml from different robots to my moveit package. However, it didn't work and I don't know which part I should configure. If I would have one default file, I could compare and find the solution faster.

I will check **moveit_config_utils** now. Thanks

reidchristopher, 2023-07-21:
@sjahr I believe I could work on this, though I definitely need help understanding the structure of things. 

What key files/classes would be I be editing? Is there an example flow of things being set/files being generated?

sjahr, 2023-07-28:
@reidchristopher Sorry for the delay. You could start by understanding how the kinematics.yaml file is generated by the setup assistant. The challenging part with the OMPL config is that you need to create the configs per group. If you look at the ompl.yaml file it has three sections:
 - [pipeline definition](https://github.com/ros-planning/moveit_resources/blob/a035b68b49cbd811e9f9595a657a47ba73dfd3b4/panda_moveit_config/config/ompl_planning.yaml#L1-L11): This is necessary to define the planner plugin that is used (OMPL) and the request adapters. It is agnostic to the robot model
 - [planner_configs](https://github.com/ros-planning/moveit_resources/blob/a035b68b49cbd811e9f9595a657a47ba73dfd3b4/panda_moveit_config/config/ompl_planning.yaml#L12-L142): Where you configure the individual planners --> This is agnostic to the robot used
 - [group-specific settings](https://github.com/ros-planning/moveit_resources/blob/a035b68b49cbd811e9f9595a657a47ba73dfd3b4/panda_moveit_config/config/ompl_planning.yaml#L144-L224): These need to be created per joint group you created with the setup assistant and you can enable/ disable certain planners here depending on your needs/ constraints

Furthermore, you need to make sure that [this section](https://github.com/ros-planning/moveit_resources/blob/a035b68b49cbd811e9f9595a657a47ba73dfd3b4/panda_moveit_config/launch/demo.launch.py#L41-L43) is part of the launch files you'd like to generate. Essentially, if you pass `ompl` as an argument for the planning_pipelines parameter, the ConfigBuilder will look in the config package for a file called `ompl`_planning.yaml and load the parameters in that file in the `ompl` namespace.

So in order to get this working you need to:
- [ ] Have MSA create an ompl_planning.yaml file in the `config` directory of the package
- [ ] That file needs to contain all three sections (1 + 2 can be copied, 3 needs to be created based on the configured joint groups)
- [ ] The MoveItConfigBuilder needs to called with `.planning_pipelines(pipelines=["ompl"])`
- [ ] Optional: It would be nice to be able to edit the ompl config in the config builder