moveit/moveit2 #2258: moveit2 ros2 humble bug in custom robot,

faheem-khaskheli, 2023-07-03:
# Setting
i am using a robot tiago ++ from webots, i exported robot as urdf, after which i have created cusotm moveit config, using setup assistant, when i run demo i give left arm a goal to reach which is valid goal, it stop working after reaching that goal.

i have made sure max_velocity and max_acceleration are float values.

# Error
[ros2_control_node-5] [INFO] [1688365040.209568727] [left_arm_controller]: Received new action goal
[ros2_control_node-5] [INFO] [1688365040.209691727] [left_arm_controller]: Accepted new action goal
[move_group-3] [INFO] [1688365040.210590729] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: left_arm_controller started execution
[move_group-3] [INFO] [1688365040.210633729] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[move_group-3] [WARN] [1688365040.236972182] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365041.246441217] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365042.246901862] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365043.266543297] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365044.276735112] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365045.277051505] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365046.287102920] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365047.323637696] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365048.356792165] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365049.386524625] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365050.416796687] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365051.456841471] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[move_group-3] [WARN] [1688365052.466747985] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: The complete state of the robot is not yet known.  Missing CASTER_WHEEL_BACK_LEFT_JOINT, SMALL_WHEEL_JOINT, CASTER_WHEEL_BACK_RIGHT_JOINT, SMALL_WHEEL_JOINT_0, CASTER_WHEEL_FRONT_LEFT_JOINT, SMALL_WHEEL_JOINT_1, CASTER_WHEEL_FRONT_RIGHT_JOINT, SMALL_WHEEL_JOINT_2, torso_lift_joint, head_1_joint, head_2_joint, wheel_left_joint, wheel_right_joint
[ros2_control_node-5] [INFO] [1688365052.656169720] [left_arm_controller]: Goal reached, success!
[move_group-3] [INFO] [1688365052.661137031] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'left_arm_controller' successfully finished
[move_group-3] [INFO] [1688365052.686752890] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-3] free(): invalid pointer
[move_group-3] Stack trace (most recent call last) in thread 3619:
[move_group-3] #15   Object "", at 0xffffffffffffffff, in 
[move_group-3] #14   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f31c6bb09ff, in 
[move_group-3] #13   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f31c6b1eb42, in 
[move_group-3] #12   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f31c6db02b2, in 
[move_group-3] #11   Object "/home/faheem/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.7.4", at 0x7f31af732d40, in move_group::MoveGroupMoveAction::executeMoveCallback(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroup> > const&)
[move_group-3] #10   Object "/home/faheem/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.7.4", at 0x7f31af7320e1, in move_group::MoveGroupMoveAction::executeMoveCallbackPlanAndExecute(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroup> > const&, std::shared_ptr<moveit_msgs::action::MoveGroup_Result_<std::allocator<void> > >&)
[move_group-3] #9    Object "/home/faheem/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.7.4", at 0x7f31af7380ac, in plan_execution::ExecutableMotionPlan::~Executab [... cut ...]
[move_group-3] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f31c6acc475, in raise
[move_group-3] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f31c6b20a7c, in pthread_kill
[move_group-3] Aborted (Signal sent by tkill() 3376 1000)
[ERROR] [move_group-3]: process has died [pid 3376, exit code -6, cmd '/home/faheem/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_rhqyce42 --params-file /tmp/launch_params_su1nw8xw'].

kappa95, 2023-07-03:
Probably is the setting up of the ROS2 controller as written here:
```bash
[ros2_control_node-5] [ERROR] [1688366470.726299793] [right_gripper_controller]: Joint name cannot be empty
[ros2_control_node-5] [WARN] [1688366470.726340793] []: Error occurred while doing error handling.
[ros2_control_node-5] [ERROR] [1688366470.726370193] [controller_manager]: After configuring, controller 'right_gripper_controller' is in state 'unconfigured' , expected inactive.
[spawner-9] [ERROR] [1688366470.743276993] [spawner_right_gripper_controller]: Failed to configure controller
[spawner-8] [INFO] [1688366470.807704193] [spawner_left_gripper_controller]: Loaded left_gripper_controller
[ros2_control_node-5] [INFO] [1688366470.814737793] [controller_manager]: Configuring controller 'left_gripper_controller'
[ros2_control_node-5] [INFO] [1688366470.814871693] [left_gripper_controller]: Action status changes will be monitored at 20Hz.
[ros2_control_node-5] [ERROR] [1688366470.814907793] [left_gripper_controller]: Joint name cannot be empty
[ros2_control_node-5] [WARN] [1688366470.814932193] []: Error occurred while doing error handling.
[ros2_control_node-5] [ERROR] [1688366470.814947793] [controller_manager]: After configuring, controller 'left_gripper_controller' is in state 'unconfigured' , expected inactive.
[spawner-8] [ERROR] [1688366470.822987693] [spawner_left_gripper_controller]: Failed to configure controller
[INFO] [spawner-6]: process has finished cleanly [pid 7943]
[INFO] [spawner-7]: process has finished cleanly [pid 7945]
[INFO] [spawner-10]: process has finished cleanly [pid 7979]
[ERROR] [spawner-9]: process has died [pid 7964, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner right_gripper_controller --ros-args'].
[ERROR] [spawner-8]: process has died [pid 7957, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner left_gripper_controller --ros-args'].
```

In Setup Assistant check the ROS2 controller section: add the FollowJointTrajectory controller for right and left group. The same name for the moveit controller. You could check the settings in the files: `ros2_controllers.yaml` and `moveit_controllers.yaml`

faheem-khaskheli, 2023-07-04:
i followed moveit setup assistant, in this for gripper there is gripper command for moveit controller, and for ros controller there is gripper action controller, i selected those.
so you are saying i have to select FollowJointTrajectory. in ros2 controller only, while moveit controller will have gripper command.

kappa95, 2023-07-04:
> i followed moveit setup assistant, in this for gripper there is gripper command for moveit controller, and for ros controller there is gripper action controller, i selected those.
> so you are saying i have to select FollowJointTrajectory. in ros2 controller. 

If you give a look on how the panda moveit config is made, you see that on the ros2 controller yaml file you have this type of controller:

      type: position_controllers/GripperActionController

Check the moveit gripper controller.yaml of the panda to have a reference of what it could be used.

faheem-khaskheli, 2023-07-04:
my files

# ros2_controllers.yaml
```
# This config file is used by ros2_control
controller_manager:
  ros__parameters:
    update_rate: 100  # Hz

    left_arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    right_arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    left_gripper_controller:
      type: position_controllers/GripperActionController

    right_gripper_controller:
      type: position_controllers/GripperActionController

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

left_arm_controller:
  ros__parameters:
    joints:
      - arm_left_1_joint
      - arm_left_2_joint
      - arm_left_3_joint
      - arm_left_4_joint
      - arm_left_5_joint
      - arm_left_6_joint
      - arm_left_7_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
right_arm_controller:
  ros__parameters:
    joints:
      - arm_right_1_joint
      - arm_right_2_joint
      - arm_right_3_joint
      - arm_right_4_joint
      - arm_right_5_joint
      - arm_right_6_joint
      - arm_right_7_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
left_gripper_controller:
  ros__parameters:
    joints:
      - left_hand_gripper_left_finger_joint
      - left_hand_gripper_right_finger_joint
right_gripper_controller:
  ros__parameters:
    joints:
      - right_hand_gripper_left_finger_joint
      - right_hand_gripper_right_finger_joint
```

# moveit_controllers.yaml
```
# MoveIt uses this configuration for controller management

moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

moveit_simple_controller_manager:
  controller_names:
    - left_arm_controller
    - right_arm_controller
    - left_gripper_controller
    - right_gripper_controller

  left_arm_controller:
    type: FollowJointTrajectory
    action_ns: follow_joint_trajectory
    default: true
    joints:
      - arm_left_1_joint
      - arm_left_2_joint
      - arm_left_3_joint
      - arm_left_4_joint
      - arm_left_5_joint
      - arm_left_6_joint
      - arm_left_7_joint
  right_arm_controller:
    type: FollowJointTrajectory
    action_ns: follow_joint_trajectory
    default: true
    joints:
      - arm_right_1_joint
      - arm_right_2_joint
      - arm_right_3_joint
      - arm_right_4_joint
      - arm_right_5_joint
      - arm_right_6_joint
      - arm_right_7_joint
  left_gripper_controller:
    type: GripperCommand
    joints:
      - left_hand_gripper_left_finger_joint
      - left_hand_gripper_right_finger_joint
    action_ns: gripper_cmd
    default: true
  right_gripper_controller:
    type: GripperCommand
    joints:
      - right_hand_gripper_left_finger_joint
      - right_hand_gripper_right_finger_joint
    action_ns: gripper_cmd
    default: true
```

kappa95, 2023-07-04:
Seems ok. But have you tried the already package for moveit for the Tiago robot? https://github.com/pal-robotics/tiago_moveit_config

faheem-khaskheli, 2023-07-04:
no, i did not try that, actually i wanted to work on new robot, so thats why i was working with setup assistant.

luisporras20, 2023-10-13:
I still have the same issue, when I execute a trajectory with "Plan and Execute" it doesn't work again
[move_group-8] [INFO] [1697226048.477924391] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'scorbot_arm_controller' successfully finished
[move_group-8] [INFO] [1697226048.478772517] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-8] free(): invalid pointer
[move_group-8] Stack trace (most recent call last) in thread 14307:
[move_group-8] #15   Object "/usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2", at 0xffffffffffffffff, in 
[move_group-8] #14   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f00a0726a3f]
[move_group-8] #13   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f00a0694ac2]
[move_group-8] #12   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f00a0adc252, in 
[move_group-8] #11   Object "/home/luis/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.8.0", at 0x7f002c199670, in move_group::MoveGroupMoveAction::executeMoveCallback(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroup> > const&)
[move_group-8] #10   Object "/home/luis/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.8.0", at 0x7f002c198e6e, in move_group::MoveGroupMoveAction::executeMoveCallbackPlanAndExecute(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroup> > const&, std::shared_ptr<moveit_msgs::action::MoveGroup_Result_<std::allocator<void> > >&)
[move_group-8] #9    Object "/home/luis/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.8.0", at 0x7f002c19e93c, in plan_execution::ExecutableMotionPlan::~ExecutableMotionPlan()
[move_group-8] #8    Object "/home/luis/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x5651e5c78c11, in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release()
[move_group-8] #7    Object "/home/luis/ws_moveit/install/moveit_ros_planning/lib/libmoveit_plan_execution.so.2.8.0", at 0x7f00a127bac0, in std::_Sp_counted_ptr_inplace<robot_trajectory::RobotTrajectory, std::allocator<robot_trajectory::RobotTrajectory>, (__gnu_cxx::_Lock_policy)2>::_M_dispose()
[move_group-8] #6    Source "./malloc/malloc.c", line 3391, in __libc_free [0x7f00a06a5452]
[move_group-8] #5    Source "./malloc/malloc.c", line 4439, in _int_free [0x7f00a06a2a43]
[move_group-8] #4    Source "./malloc/malloc.c", line 5664, in malloc_printerr [0x7f00a06a0cfb]
[move_group-8] #3    Source "../sysdeps/posix/libc_fatal.c", line 155, in __stack_chk_fail [0x7f00a0689675]
[move_group-8] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f00a06287f2]
[move_group-8] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f00a0642475]
[move_group-8] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[move_group-8]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[move_group-8]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f00a06969fc]
[move_group-8] Aborted (Signal sent by tkill() 14023 1000)
[ERROR] [move_group-8]: process has died [pid 14023, exit code -6, cmd '/home/luis/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_av8ai1t5'].

zhuxiyu24, 2024-11-11:
I got the same error. Finally, I change  the guided position_controllers/GripperActionController to joint_trajectory_controller/JointTrajectoryController for gripper. Then ,it works.

Wactor7, 2025-04-08:
> 我仍然遇到同样的问题，当我使用“计划和执行”执行轨迹时它不再起作用，但我使用的不是 Tiagos 机器人，而是我自己的机器人
> 
> [move_group-8] [INFO] [1697226048.477924391] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: 控制器“scorbot_arm_controller”已成功完成 [move_group-8] [INFO] [1697226048.478772517] [moveit_ros.trajectory_execution_manager]: 轨迹执行已完成，状态为“SUCCEEDED”... [move_group-8] free(): 无效指针 [move_group-8] 线程 14307 中的堆栈跟踪（最近一次调用最后一次）： [move_group-8] [#15](https://github.com/moveit/moveit2/issues/15) 对象“/usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2”，位于 0xffffffffffffffff，位于 [move_group-8] [#14](https://github.com/moveit/moveit2/pull/14) 源“../sysdeps/unix/sysv/linux/x86_64/clone3.S”，第 81 行，位于 _ _clone3 [0x7f00a0726a3f] [move_group-8] [#13](https://github.com/moveit/moveit2/pull/13) 源“./nptl/pthread_create.c”，第 442 行，位于 start_thread [0x7f00a0694ac2] [move_group-8] [#12](https://github.com/moveit/moveit2/pull/12) 对象“/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30”，位于 0x7f00a0adc252，位于 [move_group-8] [#11](https://github.com/moveit/moveit2/pull/11) 对象“/home/luis/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.8.0”，位于 0x7f002c199670，在 move_group::MoveGroupMoveAction::executeMoveCallback(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroup> > const&) [move_group-8] [#10](https://github.com/moveit/moveit2/pull/10) 对象“/home/luis/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.8.0”，位于 0x7f002c198e6e，在 move_group::MoveGroupMoveAction::executeMoveCallbackPlanAndExecute(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroup> > const&, std::shared_ptr<moveit_msgs::action::MoveGroup_Result_ <std::allocator > >&) [move_group-8] [#9](https://github.com/moveit/moveit2/pull/9) 对象“/home/luis/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.8.0”，位于 0x7f002c19e93c，位于 plan_execution::ExecutableMotionPlan::~ExecutableMotionPlan() [move_group-8] [#8](https://github.com/moveit/moveit2/pull/8) 对象“/home/luis/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group”，位于 0x5651e5c78c11，位于 std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() [move_group-8] [#7](https://github.com/moveit/moveit2/pull/7) 对象“/home/luis/ws_moveit/install/moveit_ros_planning/lib/libmoveit_plan_execution.so.2.8.0”，位于 0x7f00a127bac0，位于 std::_Sp_counted_ptr_inplace<robot_trajectory::RobotTrajectory，std::allocator<robot_trajectory::RobotTrajectory>，(__gnu_cxx::_Lock_policy)2>::_M_dispose() [move_group-8] [#6](https://github.com/moveit/moveit2/pull/6) 源“./malloc/malloc.c”，第 3391 行，在 __libc_free [0x7f00a06a5452] [move_group-8] [#5](https://github.com/moveit/moveit2/pull/5) 源“./malloc/malloc.c”，第 4439 行，在 _int_free [0x7f00a06a2a43] [move_group-8] [#4](https://github.com/moveit/moveit2/pull/4) 源“./malloc/malloc.c”，第 5664 行，在 malloc_printerr [0x7f00a06a0cfb] [move_group-8] [#3](https://github.com/moveit/moveit2/pull/3) 源“../sysdeps/posix/libc_fatal.c”，第 155 行，在 __stack_chk_fail [0x7f00a0689675] [move_group-8] [#2](https://github.com/moveit/moveit2/pull/2) 源“./stdlib/abort.c”，第 79 行，在 abort [0x7f00a06287f2] [move_group-8] [#1](https://github.com/moveit/moveit2/pull/1) 源“../sysdeps/posix/raise.c”，第 26 行，在 raise [0x7f00a0642475] [move_group-8] #0 | 源“./nptl/pthread_kill.c”，第 89 行，在 __pthread_kill_internal [move_group-8] |源“./nptl/pthread_kill.c”，第 78 行，在 __pthread_kill_implementation [move_group-8] 中 源“./nptl/pthread_kill.c”，第 44 行，在 __pthread_kill [0x7f00a06969fc] [move_group-8] 中止（tkill() 14023 1000 发送的信号） [错误] [move_group-8]：进程已死亡 [pid 14023，退出代码 -6，cmd'/home/luis/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_av8ai1t5']。