moveit/moveit2 #2259: [Documentation] - Improve Documentation about Cartesian Path Planners with waypoints: Bolt, Descartes, MoveIt Servo

kappa95, 2023-07-03:
### Problem Background:
Industrial tasks normally needs to move in workspace (cartesian space as preferred by most of these tutorials) more than joint space motions. 
Planning in workspace with collision avoidance is a mess, but some solutions are provided such as discussed in this [PickNik post](https://picknik.ai/cartesian%20planners/moveit/motion%20planning/2021/01/07/guide-to-cartesian-planners-in-moveit.html). But some of these solutions are not documented or mentioned (e.g., Descartes (only from ROS Industrial for ROS1), Bolt, MoveIt Servo (waypoints case if possible)). 

### A tentative solution
Add a setup of Bolt and an update of the documentation for Descartes for MoveIt2. The MoveIt Servo works for planning ahead in offline? It is allowed to be used for waypoints in such cases?

sjahr, 2023-07-12:
@kappa95 Descartes and Bolt are not yet ported to MoveIt2 which is why we don't have any documentation for this. Unfortunately, I am not aware of anybody working on this right now. Would you be interested in porting one of them over? I am happy to assist you by answering questions/ review your code. 
Besides those two, do you think there are other workspace planners that might be interesting for your use case?

sjahr, 2023-07-12:
Servo is not suitable for planning offline ahead. It can be used for online joint jogging.

kappa95, 2023-07-14:
> @kappa95 Descartes and Bolt are not yet ported to MoveIt2 which is why we don't have any documentation for this. Unfortunately, I am not aware of anybody working on this right now. Would you be [...]

I will try to study Descartes and Bolt from Moveit1 to understand better how they works and if there is something that I could try to port on my ROS2 application. Maybe it can be usefull as a starting point for a porting. 

Regarding other workspace planners I would like to use linear or polynomial representation for path in working space.

closed 2023-10-13 by github-actions[bot]

pawelir, 2025-03-14:
> [@kappa95](https://github.com/kappa95) Descartes and Bolt are not yet ported to MoveIt2 which is why we don't have any documentation for this. Unfortunately, I am not aware of anybody working on [...]

@sjahr Has anything changed referring to Descartes and Bolt ROS 2 migration? I saw this article [Guide to Cartesian Planners in MoveIt](https://picknik.ai/cartesian%20planners/moveit/motion%20planning/2021/01/07/guide-to-cartesian-planners-in-moveit.html), and in theory, the mentioned planners have significant advantages over alternatives like Pilz planner or inbuilt `computeCartesianPath()` capability.

That said, I wonder why the community doesn't see a need to port them to ROS 2. I'd appreciate spotting some light on this issue. Maybe you guys have thoughts here or know better the real performance of Descartes and Bolt compared to available planners?