moveit/moveit2 #2260: moveit2 ros2 humble error after goal execution

faheem-khaskheli, 2023-07-04:
Setting
i am using a robot tiago ++ from webots, i exported robot as urdf, after which i have created cusotm moveit config, using setup assistant.

when i run demo.launch.py, i give left arm controller or right gripper controller random valid location, after successful planning and execution, it stopped working.

# Error Logs
[ros2_control_node-5] [INFO] [1688450954.294551501] [left_arm_controller]: Goal reached, success!
[move_group-3] [INFO] [1688450954.306830003] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'left_arm_controller' successfully finished
[move_group-3] [INFO] [1688450954.335444708] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-3] free(): invalid pointer
[move_group-3] Stack trace (most recent call last) in thread 8286:
[move_group-3] #15   Object "", at 0xffffffffffffffff, in 
[move_group-3] #14   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc62037a9ff, in 
[move_group-3] #13   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc6202e8b42, in 
[move_group-3] #12   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fc62057a2b2, in 
[move_group-3] #11   Object "/home/faheem/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.7.4", at 0x7fc614f11d40, in move_group::MoveGroupMoveAction::executeMoveCallback(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroup> > const&)
[move_group-3] #10   Object "/home/faheem/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.7.4", at 0x7fc614f110e1, in move_group::MoveGroupMoveAction::executeMoveCallbackPlanAndExecute(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroup> > const&, std::shared_ptr<moveit_msgs::action::MoveGroup_Result_<std::allocator<void> > >&)
[move_group-3] #9    Object "/home/faheem/ws_moveit/install/moveit_ros_move_group/lib/libmoveit_move_group_default_capabilities.so.2.7.4", at 0x7fc614f170ac, in plan_execution::ExecutableMotionPlan::~ExecutableMotionPlan()
[move_group-3] #8    Object "/home/faheem/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x5620dbc4c9d1, in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release()
[move_group-3] #7    Object "/home/faheem/ws_moveit/install/moveit_ros_planning/lib/libmoveit_plan_execution.so.2.7.4", at 0x7fc6201d4ac0, in std::_Sp_counted_ptr_inplace<robot_trajectory::RobotTrajectory, std::allocator<robot_trajectory::RobotTrajectory>, (__gnu_cxx::_Lock_policy)2>::_M_dispose()
[move_group-3] #6    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc6202f94d2, in free
[move_group-3] #5    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc6202f6ac3, in 
[move_group-3] #4    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc6202f4d7b, in 
[move_group-3] #3    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc6202dd6f5, in 
[move_group-3] #2    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc62027c7f2, in abort
[move_group-3] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc620296475, in raise
[move_group-3] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc6202eaa7c, in pthread_kill
[move_group-3] Aborted (Signal sent by tkill() 8053 1000)
[ERROR] [move_group-3]: process has died [pid 8053, exit code -6, cmd '/home/faheem/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_zz4lh_eg --params-file /tmp/launch_params_z0x3v_53'].

faheem-khaskheli, 2023-07-04:
without changing any code i re run the demo.launch.py and this time it did not even start the execution.

# Error (second try, without modification)
[move_group-3] [INFO] [1688451194.480159532] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-3] [INFO] [1688451194.480919032] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[rviz2-4] [INFO] [1688451194.482705132] [move_group_interface]: Plan and Execute request accepted
[move_group-3] [INFO] [1688451194.495527232] [moveit_move_group_default_capabilities.move_action_capability]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[move_group-3] [INFO] [1688451194.496584332] [moveit_ros.plan_execution]: Planning attempt 1 of at most 1
[move_group-3] [INFO] [1688451194.497519832] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'ompl'
[move_group-3] [INFO] [1688451194.501146232] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'left_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-3] [WARN] [1688451195.331805030] [moveit.ros_planning.planning_pipeline]: The planner plugin did not fill out the 'planner_id' field of the MotionPlanResponse. Setting it to the planner ID name of the MotionPlanRequest assuming that the planner plugin does warn you if it does not use the requested planner.
[move_group-3] [INFO] [1688451195.332047130] [moveit.plugins.moveit_simple_controller_manager]: Returned 4 controllers in list
[move_group-3] [INFO] [1688451195.332096630] [moveit.plugins.moveit_simple_controller_manager]: Returned 4 controllers in list
[move_group-3] [INFO] [1688451195.332234330] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-3] move_group: malloc.c:4302: _int_malloc: Assertion `(unsigned long) (size) >= (unsigned long) (nb)' failed.

faheem-khaskheli, 2023-07-04:
# Error (after colcon build)
[ros2_control_node-5] [INFO] [1688458847.053762747] [left_arm_controller]: Goal reached, success!
[move_group-3] [INFO] [1688458847.086374029] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'left_arm_controller' successfully finished
[move_group-3] [INFO] [1688458847.114528514] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-3] corrupted size vs. prev_size

pinakjani, 2023-07-07:
I face a similar problem. Does it relate to improper building of move group packages?

faheemkhaskheli9, 2023-07-07:
> I face a similar problem. Does it relate to improper building of move group packages?

I fixed this issue by switching to ros2 moveit2 rolling.

sjahr, 2023-07-12:
@faheem-khaskheli Would you mind providing the full terminal output of the launch file?

sjahr, 2023-08-09:
Closing this since it seems to be solved :point_down: 

> > I face a similar problem. Does it relate to improper building of move group packages?
> 
> I fixed this issue by switching to ros2 moveit2 rolling.

closed 2023-08-09 by sjahr