moveit/moveit2 #2261: Build from source failure (moveit_py)

okvik, 2023-07-04:
### Description

Trying to build MoveIt from source following the official instructions in a fresh (and updated) Humble container I get the error about `pybind11` include path. The full build output of the `moveit_py` package is below, the interesting part being:

```
-- Found pybind11: //include (found version "2.9.1")
```

It's been a while since I've built MoveIt from source so I don't know when this error might have started occurring. 

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04 (Docker)
* Source build: `main` branch as of this moment
* Which RMW: CycloneDDS

### Steps to reproduce

This is basically the Dockerfile I'm using. The manual part is simply following https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html almost to a word. I've got several of my own packages in the same workspace but I doubt this matters.

```
FROM osrf/ros:humble-desktop-full

RUN apt-get update
RUN apt-get install -y \
    tmux \
    ccache \
    ninja-build \
    clangd \
    ripgrep

RUN apt-get install -y \
    ros-humble-rmw-cyclonedds-cpp \
    ros-humble-rqt \
    ros-humble-rqt-* \
    ros-humble-graph-msgs
```

### Backtrace or Console output
```
Starting >>> moveit_py
ninja: error: loading 'build.ninja': No such file or directory
--- output: moveit_py
Not searching for unused variables given on the command line.
-- Found ament_cmake: 1.3.5 (/opt/ros/humble/share/ament_cmake/cmake)
-- Found Python3: /bin/python3.10 (found version "3.10.6") found components: Interpreter 
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rclcpp: 16.0.4 (/opt/ros/humble/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.1.4 (/opt/ros/humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.4 (/opt/ros/humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.4 (/opt/ros/humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found rmw_implementation_cmake: 6.1.1 (/opt/ros/humble/share/rmw_implementation_cmake/cmake)
-- Found rmw_fastrtps_cpp: 6.2.2 (/opt/ros/humble/share/rmw_fastrtps_cpp/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rclpy: 3.3.8 (/opt/ros/humble/share/rclpy/cmake)
-- Found geometry_msgs: 4.2.3 (/opt/ros/humble/share/geometry_msgs/cmake)
-- Found moveit_ros_planning: 2.7.4 (/home/kvik/src/robot/install/moveit_ros_planning/share/moveit_ros_planning/cmake)
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: system filesystem date_time program_options thread chrono 
-- Found parameter_traits: 0.3.3 (/opt/ros/humble/share/parameter_traits/cmake)
-- Found rclcpp_lifecycle: 16.0.4 (/opt/ros/humble/share/rclcpp_lifecycle/cmake)
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: chrono date_time filesystem iostreams program_options regex serialization system thread 
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: filesystem 
-- library: /usr/lib/x86_64-linux-gnu/libcurl.so
-- Ensuring Eigen3 include directory is part of orocos-kdl CMake target
-- Found moveit_ros_planning_interface: 2.7.4 (/home/kvik/src/adapto/ekos/robot/install/moveit_ros_planning_interface/share/moveit_ros_planning_interface/cmake)
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: date_time filesystem program_options system thread 
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: thread system filesystem regex date_time program_options 
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: system filesystem date_time program_options thread 
-- Found Python3: /bin/python3.10 (found version "3.10.6") found components: Interpreter Development Development.Module Development.Embed 
-- Found pybind11_vendor: 2.4.2 (/opt/ros/humble/share/pybind11_vendor/cmake)
-- Found pybind11: //include (found version "2.9.1")
-- Found ament_cmake_pytest: 1.3.5 (/opt/ros/humble/share/ament_cmake_pytest/cmake)
-- Configuring done
CMake Error in CMakeLists.txt:
  Imported target "pybind11::headers" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "pybind11::headers" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

sjahr, 2023-07-12:
:thinking: @peterdavidfagan Do you have an idea what the issue could be? @okvik Could you try testing if the issue still persists if you follow the instructions to setup a docker with moveit2 https://moveit.picknik.ai/main/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html?

sjahr, 2023-07-12:
Awesome, I am closing this issue then :+1:

closed 2023-07-12 by sjahr