moveit/moveit2 #2265: OMPL <unspecified> | failed trajectory execution

KlausLex, 2023-07-12:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Humble]
* OS Version: e.g. Ubuntu 22.04
* Binary build (2.5.4-1jammy.20230518.202430)
* Tested using Fast DDS & Cyclone DDS
* Running inside Docker

### Steps to reproduce
* Use [this xacro](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/ros2/urdf/ur.urdf.xacro) to create an URDF for a simple ur10e with no tool or environment attached
* Create MoveIt 2 config via MSA & launch **demo.launch.py** & **move_group.launch.py**

### Expected behaviour
1) Set goal state & move robot clicking **Plan & Execute**.
2) Selecting different planners from OMPL.

### Actual behaviour
1) Clicking planning & executing loops the planner visualization endlessly & never executes.
2) OMPL dropdown is empty displaying <unspecified>

### Console output **demo.launch.py**

```
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [static_transform_publisher-1]: process started with pid [841]
[INFO] [robot_state_publisher-2]: process started with pid [843]
[INFO] [move_group-3]: process started with pid [845]
[INFO] [rviz2-4]: process started with pid [847]
[INFO] [ros2_control_node-5]: process started with pid [849]
[INFO] [spawner-6]: process started with pid [861]
[INFO] [spawner-7]: process started with pid [870]
[static_transform_publisher-1] [INFO] [1689159227.593508918] [static_transform_publisher0]: Spinning until stopped - publishing transform
[static_transform_publisher-1] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-1] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-1] from 'world' to 'base_link'
[rviz2-4] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[robot_state_publisher-2] [INFO] [1689159227.767741663] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1689159227.768959852] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1689159227.770029704] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-2] [INFO] [1689159227.770920410] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1689159227.771795905] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-2] [INFO] [1689159227.772661799] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-2] [INFO] [1689159227.773490381] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-2] [INFO] [1689159227.774402627] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1689159227.775365990] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-2] [INFO] [1689159227.776351010] [robot_state_publisher]: got segment world
[robot_state_publisher-2] [INFO] [1689159227.777315733] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-2] [INFO] [1689159227.777340364] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-2] [INFO] [1689159227.777354637] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-5] text not specified in the script_filename tag
[ros2_control_node-5] text not specified in the output_recipe_filename tag
[ros2_control_node-5] text not specified in the input_recipe_filename tag
[ros2_control_node-5] text not specified in the tf_prefix tag
[ros2_control_node-5] [INFO] [1689159227.808073264] [resource_manager]: Loading hardware 'ur10e' 
[ros2_control_node-5] [INFO] [1689159227.815744638] [resource_manager]: Initialize hardware 'ur10e' 
[ros2_control_node-5] [INFO] [1689159227.815907042] [resource_manager]: Successful initialization of hardware 'ur10e'
[ros2_control_node-5] [INFO] [1689159227.816144483] [resource_manager]: Loading hardware 'FakeSystem' 
[ros2_control_node-5] [INFO] [1689159227.819588829] [resource_manager]: Initialize hardware 'FakeSystem' 
[ros2_control_node-5] [WARN] [1689159227.820961406] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-5] <state_interface name="velocity"> 
[ros2_control_node-5]   <param name="initial_value">0.0</param> 
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1689159227.824117467] [resource_manager]: Successful initialization of hardware 'FakeSystem'
[ros2_control_node-5] [INFO] [1689159227.825606194] [resource_manager]: 'configure' hardware 'ur10e' 
[ros2_control_node-5] [INFO] [1689159227.826897342] [resource_manager]: Successful 'configure' of hardware 'ur10e'
[ros2_control_node-5] [INFO] [1689159227.830161050] [resource_manager]: 'activate' hardware 'ur10e' 
[... 343 lines cut ...]
[move_group-1] [INFO] [1689159292.375839783] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to manipulator_controller
[move_group-1] [INFO] [1689159292.377908108] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: manipulator_controller started execution
[move_group-1] [INFO] [1689159292.377931382] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[move_group-1] [INFO] [1689159299.927907635] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'manipulator_controller' successfully finished
[move_group-1] [INFO] [1689159299.954979518] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-1] [INFO] [1689159299.955317734] [moveit_move_group_default_capabilities.move_action_capability]: Solution was found and executed.
```
* When switching to Cyclone DDS, **move_group.launch.py** prints this when **demo.launch.py** is running:
```
[move_group-1] 2023-07-12 09:02:39.750 [RTPS_READER_HISTORY Error] Change payload size of '36' bytes is larger than the history payload size of '23' bytes and cannot be resized. -> Function can_change_be_added_nts
[move_group-1] 2023-07-12 09:02:39.846 [RTPS_READER_HISTORY Error] Change payload size of '32' bytes is larger than the history payload size of '19' bytes and cannot be resized. -> Function can_change_be_added_nts
```
* There is an error `[ros2_control_node-4] [ERROR] [1689152766.956593400] [UR_Client_Library]: Opening file '' failed with error: No such file or directory` but the package is called **Universal_Robots_Client_Library**?
* There is an error `[ros2_control_node-4] [FATAL] [1689152766.956693548] [URPositionHardwareInterface]: Opening file '' failed with error: No such file or directory` but isn't this the **Universal_Robots_ROS2_Driver**?

### Questions
* How do I create a working MoveIt 2 configuration for ROS 2 Humble & access planners from OMPL for a real UR10e?
* Since the default configuration shipped with the UR_Driver works for me, can we reuse it by just replacing the robot definition with collisions generated from MSA if we have static end-effectors and no urdf for the environment?
* Is it necessary to define end-effectors & respective planning groups when they're static?
* Are we not supposed to use **demo.launch.py**? I figured this is self-containing & just for validating the generated config?
* The folder contents for /config & /launch looks different to other projects I've seen, such as ros2_control.xacro has a name prefix and there is no ompl_planning.yaml or controllers.yaml.
* I'm not sure as what to make of this, but it's kinda hard for me to track down what version to use?
* Looking at other issues here, Humble doesn't seem too stable?

Thanks & have a great day!

sjahr, 2023-07-13:
Hi @KlausLex,
for issues with the UR driver please open an issue in that repository. 

For your MoveIt issues: Make sure you've installed/ build MoveIt2 correctly following the [instructions on the website](https://moveit.ros.org/install-moveit2/source/) or the [docker tutorial](https://moveit.picknik.ai/main/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html?highlight=docker). Regarding any missing config files or issues you can compare your MoveIt config to an [existing configs](https://github.com/ros-planning/moveit_resources) to see what might be missing/ different.

Regarding your questions: Taking a look at the [tutorials](https://moveit.picknik.ai/main/index.html) and [concepts](https://moveit.ros.org/documentation/concepts/) page might help you to get answers. Feel free to ask remaining questions but please be more respectful of other people's time and try to keep the number of questions small and precise, so it is easy to help you.

KlausLex, 2023-07-14:
Thanks for your reply & sorry for taking up so much of your time @sjahr .

My questions essentially boils down to a single one:
**What extra steps are necessary after using MSA 2.0 to achieve the same configuration state as MSA 1.0?**

As I mentioned in my edits, with the same URDF I can generate 2 config packages by pulling the **official docker images**:
### **_[noetic-source](https://hub.docker.com/layers/moveit/moveit/noetic-source/images/sha256-e58035ce6c55c9412dfd1e5b0663f9ce46885809a05f29b73cf6644108cadef5?context=explore)_**
generates a config package with **OMPL**, **STOMP**, **CHOMP** & **Pilz** planners I can run with **demo.launch**
### **_[humble-source](https://hub.docker.com/layers/moveit/moveit2/humble-source/images/sha256-a8f324ae966205be5efc16f2cd2bda7d011a5c2661255d7e21e07d7d407d31c0?context=explore)_**
generates a config package with **STOMP**, **CHOMP** & **Pilz** planners I cannot run with **demo.launch.py** & **OMPL staying at "<"unspecified">"** even after manually creating **ompl_planning.yaml** & adding OMPL to the **planning_pipeline** in a new launch-file and also installing OMPL via apt.
### Installing **_ros-humble-moveit_ via APT**
generates a config package **with Pilz planners only** that I cannot run with **demo.launch.py** either.
### **_building MoveIt 2 from source_**
had 2 of computers I have access to crash & reboot on building (1 laptop with 8GB of memory & one desktop PC with 24GB of memory)

Following the [MSA 2.0 Example](https://moveit.picknik.ai/main/doc/examples/setup_assistant/setup_assistant_tutorial.html) using the **_docker image_** **[humble-source](https://hub.docker.com/layers/moveit/moveit2/humble-source/images/sha256-a8f324ae966205be5efc16f2cd2bda7d011a5c2661255d7e21e07d7d407d31c0?context=explore)** generates a **panda_moveit_config package** that is very different from the one in the [moveit_resources repo](https://github.com/ros-planning/moveit_resources/tree/humble) with *_planning.yaml missing (incl. OMPL), as well as contents of **demo.launch.py** differ greatly.
So I don't think **panda_moveit_config** inside **moveit_resources** repo was created with any of the currently available MSA 2.0?

_I may have possibly overlooked where the docs mention the outcome from MSA 2.0 differs from MSA 1.0, or MSA 2.0 not creating packages similar to the ones inside **moveit_resources** & would be greatful for being pointed into the right direction._

sjahr, 2023-07-17:
Just adding the ompl_planning.yaml file is not sufficient. It requires to contain the following lines (see [this file as reference](https://github.com/ros-planning/moveit_resources/blob/humble/panda_moveit_config/config/ompl_planning.yaml)):

```
planning_plugin: ompl_interface/OMPLPlanner
request_adapters: >-
  default_planner_request_adapters/AddTimeOptimalParameterization
  default_planner_request_adapters/ResolveConstraintFrames
  default_planner_request_adapters/FixWorkspaceBounds
  default_planner_request_adapters/FixStartStateBounds
  default_planner_request_adapters/FixStartStateCollision
  default_planner_request_adapters/FixStartStatePathConstraints
 ```
and you need to make sure that is loaded. The MoveItConfig in the demo.launch.py file needs to contain the pipeline names to be loaded

```
    moveit_config = (
        MoveItConfigsBuilder("moveit_resources_panda")
        .robot_description(
            file_path="config/panda.urdf.xacro",
            mappings={
                "ros2_control_hardware_type": LaunchConfiguration(
                    "ros2_control_hardware_type"
                )
            },
        )
        .robot_description_semantic(file_path="config/panda.srdf")
        .trajectory_execution(file_path="config/gripper_moveit_controllers.yaml")
        .planning_pipelines(
            pipelines=["ompl", "chomp", "pilz_industrial_motion_planner"]
        )
        .to_moveit_configs()
    )
 ```
 The `.planning_pipelines(pipelines=["ompl", "chomp", "pilz_industrial_motion_planner"])` is the relevant part here. For each item in this list, the config class will look for a file called PIPELINENAME_planning.yaml in the config directory and load these parameters. When you initialize a move_group, MoveIt2 will look up the pipeline names in the I think `planning_pipelines` parameter and for each name in that list if there is a corresponding namespace a pipeline is loaded if that namespace contains a `planning_plugin` parameter and the parameters required by the planning plugin.

h-matatagi, 2023-08-04:
Hi, I'm having a similar issue with the Pilz Industrial Motion Planner. My pilz_industrial_motion_planner_planning.yaml is the same as the [file available in moveit_resources](https://github.com/ros-planning/moveit_resources/blob/humble/panda_moveit_config/config/pilz_industrial_motion_planner_planning.yaml), but the planner is still <unspecified> even though the yaml file clearly specifies the default planner as PTP.

The MoveItConfig in my launch file is also very similar to demo.launch.py. I have `.planning_pipelines("pilz_industrial_motion_planner", ["pilz_industrial_motion_planner", "chomp", "ompl"])`.

Is there something wrong with the RViz visualization? Or am I missing something? Thank you in advance!

closed 2023-11-29 by github-actions[bot]