moveit/moveit2 #2272: Add utility function to check trajectory continuity

sjahr, 2023-07-27:
To verify the result of a motion planner/generator it might be necessary to check a trajectory for joint jumps. It would be nice to have this utility available as a free function. There is already [an implementation available in the Cartesian interpolator](https://github.com/ros-planning/moveit2/blob/main/moveit_core/robot_state/src/cartesian_interpolator.cpp#L222) that needs to be refactored into a free function. For more context take a look at [this (unmerged PR)](https://github.com/ros-planning/moveit2/pull/2253) with a less mature implementation.

abhijelly, 2023-08-22:
hi @sjahr, can I work on this?

EzraBrooks, 2023-10-10:
not stale.

@abhijelly thanks for the offer (and sorry that nobody followed up with you here)! PickNik has a commitment to deliver this particular feature to a client soon so we'll be handling the implementation to ensure it meets those requirements.

closed 2023-11-07 by henningkayser