moveit/moveit2 #2275: Break out parameters for `servo publish rate` and `trajectory point duration` for improved servoing via position control

nbbrooks, 2023-07-27:
# Summary
Certain robot arms that only have a position control interface (e.g. no velocity control support) experience non-smooth and noisy (bad noises from gearbox) servo performance.

My current hypothesis is that this is related to using the same time parameter for both [publish period loop](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/src/servo_calcs.cpp#L314) and the [published JointTrajectoryPoint duration](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/src/servo_calcs.cpp#L195). Servoing via a position controller could cause the below behavior, where the arm's onboard control works aggressively works to de-accelerate and be at 0 velocity at the end of the JointTrajectoryPoint duration, then it immediately receives a new JointTrajectoryPoint telling it to accelerate again.

[image]

# Proposed change
We should try making these parameters separate, so we do the following and test if it resolves the poor performance on certain hardware. We should also enforce that your trajectory point duration equal to or shorter than your publish duration.

[image]

# Describe alternatives you've considered
- Make no changes and document requiresments for arm manufacturers around either position control filtering techniques and/or requiring a velocity control interface.
- Make no changes and instead use onboard jogging services from these arm manufacturers (if they exist).
  - We lose our collision checking here.

pac48, 2023-07-27:
I believe ros2_cotrol  allows for setting endpoint velocities even if the robot is not velocity controlled. The default interpolation is cubic for a reference with position and velocity values. I ran a quick simulation to show how interpolation affects the spline's shape. 
 
[image]

The velocity profile @nbbrooks highlighted occurs when the endpoint velocities of the reference are 0. But I believe if they are set to non-zero you won't have the "noise" part.  This needs further validation, however.

nbbrooks, 2023-07-28:
Thanks @AndyZe - glad your intuition follows here.

Mostly curious here, but do you remember what ended up being the Gz issue you referenced in #1812 ?

AndyZe, 2023-07-28:
The Gazebo issue was a slow sag over time, with gravity. I believe that's been fixed by a PR to gz_ros2_control now.

What's probably more interesting for you is that https://github.com/ros-planning/moveit2/pull/886 fixed a similar issue to what you described for a UR robot. This was when I came out to Boulder for the week to troubleshoot. The solution was a version of "moving the carrot farther ahead" so there's do sudden discontinuity.

It's a pain point for me that the PR wasn't merged considering it was verified to fix the issue on hardware and Gijs vdH said it jived with what some robot manufacturers do.

JaisonJose241, 2023-08-20:
> @pac48 you also have to think about what the low-level robot controllers are capable of
> 
> I believe this issue can be tuned out by setting a Servo rate that is faster than the ros2_control rate so there is overlap between commands.
> 
> Related MR: #1812

Tried the same on hardware ur5, and it shows no movement, while it works perfectly with jointgroup*contorllers.

moriarty, 2023-08-29:
> Tried the same on hardware ur5

Yes this issue wasn't specific when it mentioned certain robot arms

> Certain robot arms

Some robot hardware seems to be doing internal filtering and some doesn't filter.

This is seen on the Kinova https://github.com/ros-planning/moveit2/issues/2275 and maybe on the Franka Panda (ref needed)

JaisonJose241, 2023-08-29:
Is their any work around?

sea-bass, 2024-01-18:
I think this issue has been superseded by @pac48's work in https://github.com/ros-planning/moveit2/pull/2594, so will close this one.

closed 2024-01-18 by sea-bass