moveit/moveit2 #2280: humble-ci-testing + clang-tidy: Gmock deprecation

henningkayser, 2023-08-01:
### Description

CI is failing due to a deprecation in gmock [here](https://github.com/ros-planning/moveit2/actions/runs/5715687575/job/15485573319?pr=2277#step:12:5040)

```
In file included from /opt/ros/humble/src/gmock_vendor/src/gmock_main.cc:32:
In file included from /opt/ros/humble/src/gmock_vendor/include/gmock/gmock.h:61:
In file included from /opt/ros/humble/src/gmock_vendor/include/gmock/gmock-function-mocker.h:39:
In file included from /opt/ros/humble/src/gmock_vendor/include/gmock/gmock-generated-function-mockers.h:47:
In file included from /opt/ros/humble/src/gmock_vendor/include/gmock/gmock-spec-builders.h:75:
/opt/ros/humble/src/gmock_vendor/include/gmock/gmock-matchers.h:721:3: error: definition of implicit copy constructor for 'StrEqualityMatcher<std::basic_string<char>>' is deprecated because it has a user-declared copy assignment operator [-Werror,-Wdeprecated-copy]
  GTEST_DISALLOW_ASSIGN_(StrEqualityMatcher);
  ^ 
/opt/ros/humble/src/gtest_vendor/include/gtest/internal/gtest-port.h:682:8: note: expanded from macro 'GTEST_DISALLOW_ASSIGN_'
  void operator=(type const &) = delete
       ^ 
/opt/ros/humble/src/gtest_vendor/include/gtest/gtest-matchers.h:456:54: note: in implicit copy constructor for 'testing::internal::StrEqualityMatcher<std::basic_string<char>>' first required here
  explicit PolymorphicMatcher(const Impl& an_impl) : impl_(an_impl) {}
                                                     ^
/opt/ros/humble/src/gtest_vendor/include/gtest/gtest-matchers.h:514:10: note: in instantiation of member function 'testing::PolymorphicMatcher<testing::internal::StrEqualityMatcher<std::basic_string<char>>>::PolymorphicMatcher' requested here
  return PolymorphicMatcher<Impl>(impl);
         ^ 
/opt/ros/humble/src/gmock_vendor/include/gmock/gmock-matchers.h:3813:10: note: in instantiation of function template specialization 'testing::MakePolymorphicMatcher<testing::internal::StrEqualityMatcher<std::basic_string<char>>>' requested here
  return MakePolymorphicMatcher(

```

ChrisThrasher, 2023-08-01:
I fixed this elsewhere by disable `-Wdeprecated-copy`. This is our best short term solution.

henningkayser, 2023-08-02:
@ChrisThrasher thanks, yeah I thought something like that would be the best solution right now. The original issues have been fixed long ago in gmock/googletest, but we're still somewhat stuck on 1.10 from 2019 because of the ROS Humble. At least, Rolling is using 1.11 since https://github.com/ament/ament_cmake/pull/408, fixing these exact issues. I'll add a PR for disabling the check, anyway.

henningkayser, 2023-08-02:
Wait... we actually used to run on gtest-1.11 in CI using [this PR branch](https://github.com/ros-planning/moveit2/blob/humble/.github/workflows/ci.yaml#L28), which must have been removed by GitHub recently.

https://github.com/ros-planning/moveit2/pull/2283 fixes the installed version for the Humble branch, https://github.com/ros-planning/moveit2/pull/2284 removes the redundant and silently failing install from main.

PR #2283 (2023-08-02): Humble: Use googletest 1.11.9000
Fixes #2280.
The installed PR branch has been removed by GitHub, so CI silently switched to the official ament/googletest release at version 1.10 producing compiler warnings. This fixes the version to a more recent one by installing a later release tag of the vendor package directly.