moveit/moveit2 #2289: can not make the panda move when i click 'plan & execute'

Zhaogaog, 2023-08-07:
I’m using ubuntu22.04,ros2humble, and when I follow this tutorial([9. MoveIt 2 — isaacsim latest documentation](https://docs.omniverse.nvidia.com/isaacsim/latest/tutorial_ros2_moveit.html))、([How To Command Simulated Isaac Robot — MoveIt Documentation: Rolling documentation](https://moveit.picknik.ai/main/doc/how_to_guides/isaac_panda/isaac_panda_tutorial.html)), I have a couple problems:
First of all: “controller manager not available”,All I can ensure is that I have the controller manager installed. And the ‘[ERROR] [move_group-5]: process has died [pid 59, exit code -6, cmd ‘/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --log-level info --ros-args --params-file /tmp/launch_params_89dev4yd’].’ appears.Besides,I can not see ’
You can start planning now!’ in the terminal.
Secondly: in the rviz2 interface, there is a warning like 'requesting initial scene failed '.
Lastly, when I move the trackball and give control commands, the arm does not move when I click ‘plan & execute’ and I get this error in the terminal 'Movegroup action client/server not ready '.
Help ！！！！！！

Screenshot from 2023-08-05 17-21-49
Screenshot from 2023-08-05 17-21-49
1920×1078 505 KB

Screenshot from 2023-08-05 17-26-21
Screenshot from 2023-08-05 17-26-21
1920×1078 216 KB

Screenshot from 2023-08-05 17-26-25
Screenshot from 2023-08-05 17-26-25
1920×1078 422 KB

sjahr, 2023-08-09:
Duplicate of #2290

closed 2023-08-09 by sjahr

yangming517, 2024-03-12:
hello ,i wanna know that if you have solve this problem? I have the same problem now!