moveit/moveit2 #2291: Load robot_description from topic

Noel215, 2023-08-10:
Hi,

I would like to be able to load the robot description from the `/robot_description` topic instead of passing it as a parameter. 
This feature would be great in order to centralize the way of getting the description for all nodes. Right now I need to load the robot description and do the proper substitutions on each different launcher I have, as in [tiago_moveit_config/launch/move_group.launch.py](https://github.com/pal-robotics/tiago_moveit_config/blob/humble-devel/launch/move_group.launch.py#L57-L67), [tiago_moveit_config/launch/moveit_rviz.launch.py](https://github.com/pal-robotics/tiago_moveit_config/blob/humble-devel/launch/moveit_rviz.launch.py#L57-L67) and the topic, which sometimes leads to different robot descriptions.

I'm not really sure if that's possible with the current implementation but I've seen some other issues mentioning it as https://github.com/ros-planning/moveit2/issues/169, https://github.com/ros-planning/moveit2/issues/755, https://github.com/ros-planning/moveit2/issues/1511, among others. I've also found that the node is able to publish the robot description, but nothing about reading it from a topic.

Could you please let me know if that feature is available right now? 
If not, is it planned to add soon to MoveIt 2?

Thanks :)

Noel215, 2023-09-12:
Hi,

Could I have a response on this, please?

saikishor, 2023-11-07:
Hello!

We would like to have this feature on our robots. Is it possible to have this in the near future?

Thank you,

henningkayser, 2024-02-27:
This has already been implemented a while ago in https://github.com/ros-planning/moveit2/pull/765, and since https://github.com/ros-planning/moveit2/pull/1806 all you need to do is leave the robot_description parameter empty in your launch file, so that MoveIt loads it from `/robot_description`. I'm closing this issue. If you would improve the documentation in a way that would have prevented you from filing this issue, that would be greatly appreciated.

closed 2024-02-27 by henningkayser