moveit/moveit2 #2294: unreferenced boost dependencies (libmoveit_distance_field.so.2.5.4: undefined reference to `boost::iostreams::zlib::sync_flush')

SirTobias, 2023-08-11:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04 VM from os.boxes.org
* Source build from [instructions](https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html)
* If source, which branch?
* Which RMW: Cyclone DDS

### Steps to reproduce
`mkdir -p ~/ros2_moveit/src`
`cd ~/ros2_moveit/src`
`pip3 install colcon-mixin`
`sudo apt install librange-v3-dev`

Checkout MoveIt2 for Humble:

`git clone https://github.com/ros-planning/moveit2_tutorials -b humble --depth 1`
`vcs import < moveit2_tutorials/moveit2_tutorials.repos`
`sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`
`cd ~/ros2_moveit`
`colcon build --mixin release`
Missing repos for MoveIt2, I do not know, why they are not included in moveit2_tutorials.repos` :

`git clone git@github.com:ros-planning/moveit_msgs.git -b ros2`
`git clone git@github.com:ros-planning/moveit_resources.git -b humble`
`git clone git@github.com:wg-perception/object_recognition_msgs.git -b ros2`
`git clone git@github.com:OctoMap/octomap_msgs.git -b ros2`
`git clone git@github.com:PickNikRobotics/generate_parameter_library.git`
`git clone git@github.com:PickNikRobotics/RSL.git`
`git clone git@github.com:PickNikRobotics/cpp_polyfills.git`
`git clone git@github.com:pantor/ruckig.git`

`colcon build --mixin release`

### Expected behaviour
MoveIt2 should be compiled

### Actual behaviour
Undefined reference to boost::iostreams::zlib::sync_flush
`apt search libboost-all-dev`
gives
Sorting... Done
Full Text Search... Done
libboost-all-dev/jammy,now 1.74.0.3ubuntu7 amd64 [installed,automatic]
  Boost C++ Libraries development files (ALL) (default version)

### Backtrace or Console output

[Console Output](https://gist.github.com/SirTobias/d3e6497bd4b0fb30754e6164afa47ad4)

Thank you very much for your help in advance.

tmayoff, 2023-10-17:
This seems to be an issue with clang, @SirTobias, not sure if you're still encountering this, but any chance you're using clang to compile it. When I switched to gcc it no longer failed at that point. It now fails elsewhere but still progress

closed 2024-02-20 by moriarty

yyxmoon123, 2024-12-03:
I encountered the same error.