moveit/moveit2 #2334: Unable to plan to desired pose

Giovadess, 2023-08-29:
### Description
Hello, I am currently working with the stretch roboto demo, for planning for a mobile base manipulator.
I was able to run everything smoothly and now I started implementing the Cpp API.
Currently I am able to plan both using the Rviz interface, and using a simple implementation in the Cpp API sending a goal in the joint space such as:
  joint_group_positions[0] = 1;   // 0 to inf, position/x
  joint_group_positions[1] = 0;    // 0 to inf, position/y
  joint_group_positions[2] = 0.5;  // 0 to 3.14, position/theta
  move_group_base_arm.setJointValueTarget(joint_group_positions);
Both of this work fine.
The problem comes when I try to set a goal pose in the cartesian space for my planning group. I tried something like this:

`  static const std::string PLANNING_GROUP_MOBILE_BASE_ARM = "mobile_base_arm";
  moveit::planning_interface::MoveGroupInterface move_group_base_arm(move_group_node, PLANNING_GROUP_MOBILE_BASE_ARM);

  // Set the target pose (position and orientation) for the end effector
  geometry_msgs::msg::Pose target_pose;
  // target_pose.frame_id= "odom";
  target_pose.position.x = 0.6;    // Set your desired x position
  target_pose.position.y = 0.6;    // Set your desired y position
  target_pose.position.z = 0.0;    // Set your desired z position

  target_pose.orientation.x = 0.0; // Set your desired orientation quaternion
  target_pose.orientation.y = 0.0;
  target_pose.orientation.z = 0.0;
  target_pose.orientation.w = 1.0;
  // Set the target pose
  move_group_base_arm.setPoseTarget(target_pose);`

I have tried this both with the stretch robot and the mobile base I am working with both set up with the stretch_kinematics_plugin. The planner always fails, with a message of this type.
[image]

Sorry, for the edit I posted the wrong issue initially.

### Your environment
* ROS Distro: [Galactic]
* OS Version: e.g. Ubuntu 20.04

### Expected behaviour

Plan to a desired pose using the Cpp API.

### Actual behaviour
Plan fails 
[image]

## Error message

[INFO] [launch]: All log files can be found below /home/gdessy/.ros/log/2023-08-29-10-44-50-602933-iitavr005dw002u-16830
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [16839]
[INFO] [robot_state_publisher-2]: process started with pid [16841]
[INFO] [move_group-3]: process started with pid [16843]
[INFO] [rviz2-4]: process started with pid [16845]
[INFO] [ros2_control_node-5]: process started with pid [16847]
[INFO] [spawner-6]: process started with pid [16849]
[INFO] [spawner-7]: process started with pid [16851]
[static_transform_publisher-1] [INFO] [1693298692.122746703] [static_transform_publisher]: Spinning until killed publishing transform from 'odom' to 'base_link'
[robot_state_publisher-2] The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-2] Link caster_link had 0 children
[robot_state_publisher-2] Link link_aruco_left_base had 0 children
[robot_state_publisher-2] Link link_aruco_right_base had 0 children
[robot_state_publisher-2] Link link_left_wheel had 0 children
[robot_state_publisher-2] Link link_mast had 3 children
[robot_state_publisher-2] Link link_head had 1 children
[robot_state_publisher-2] Link link_head_pan had 1 children
[robot_state_publisher-2] Link link_head_tilt had 0 children
[robot_state_publisher-2] Link link_lift had 2 children
[robot_state_publisher-2] Link link_arm_l4 had 1 children
[robot_state_publisher-2] Link link_arm_l3 had 1 children
[robot_state_publisher-2] Link link_arm_l2 had 1 children
[... 245 lines cut ...]
[move_group-3] 
[move_group-3] [INFO] [1693298709.190480077] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[move_group-3] [INFO] [1693298709.190493879] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context initialization complete
[move_group-3] Loading 'move_group/ApplyPlanningSceneService'...
[move_group-3] Loading 'move_group/ClearOctomapService'...
[move_group-3] Loading 'move_group/MoveGroupCartesianPathService'...
[move_group-3] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-3] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[move_group-3] Loading 'move_group/MoveGroupKinematicsService'...
[move_group-3] Loading 'move_group/MoveGroupMoveAction'...
[move_group-3] Loading 'move_group/MoveGroupPlanService'...
[move_group-3] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-3] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-3] 
[move_group-3] You can start planning now!
[move_group-3] 
[rviz2-4] [INFO] [1693298736.286525156] [interactive_marker_display_94791558449984]: Sending request for interactive markers
[rviz2-4] [INFO] [1693298736.320862128] [interactive_marker_display_94791558449984]: Service response received for initialization
[move_group-3] [INFO] [1693298741.758034821] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-3] [INFO] [1693298741.758244525] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-3] [INFO] [1693298741.765394606] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-3] [INFO] [1693298741.766929992] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'mobile_base_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-3] [ERROR] [1693298746.768006690] [ompl]: /tmp/binarydeb/ros-galactic-ompl-1.5.2/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:253 - mobile_base_arm/mobile_base_arm: Unable to sample any valid states for goal tree
[move_group-3] [INFO] [1693298747.034230082] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-3] [INFO] [1693298747.034328984] [moveit_move_group_default_capabilities.move_action_capability]: No motion plan found. No execution attempted.

rhaschke, 2023-08-29:
This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here only.

To give a very short reply on your question: The key error message is:
> `mobile_base_arm/mobile_base_arm: Unable to sample any valid states for goal tree`

Are you sure your robot can reach the given pose, particularly the identity orientation?

> It seems that my Inverse kinematics is not instantiated.

You would get a clear error message on the console about this, but I haven't seen anything in the log.

closed 2023-08-29 by rhaschke

relifeto18, 2024-02-07:
Hi, I'm facing the same problem, did you solve it?

> ### Description
> Hello, I am currently working with the stretch roboto demo, for planning for a mobile base manipulator. I was able to run everything smoothly and now I started implementing the Cpp API. Currently I am able to plan both using the Rviz interface, and using a simple implementation in the Cpp API sending a goal in the joint space such as: joint_group_positions[0] = 1; // 0 to inf, position/x joint_group_positions[1] = 0; // 0 to inf, position/y joint_group_positions[2] = 0.5; // 0 to 3.14, position/theta move_group_base_arm.setJointValueTarget(joint_group_positions); Both of this work fine. The problem comes when I try to set a goal pose in the cartesian space for my planning group. I tried something like this:
> 
> ```c++
>   static const std::string PLANNING_GROUP_MOBILE_BASE_ARM = "mobile_base_arm";
>   moveit::planning_interface::MoveGroupInterface move_group_base_arm(move_group_node, PLANNING_GROUP_MOBILE_BASE_ARM);
> 
>   // Set the target pose (position and orientation) for the end effector
>   geometry_msgs::msg::Pose target_pose;
>   // target_pose.frame_id= "odom";
>   target_pose.position.x = 0.6;    // Set your desired x position
>   target_pose.position.y = 0.6;    // Set your desired y position
>   target_pose.position.z = 0.0;    // Set your desired z position
> 
>   target_pose.orientation.x = 0.0; // Set your desired orientation quaternion
>   target_pose.orientation.y = 0.0;
>   target_pose.orientation.z = 0.0;
>   target_pose.orientation.w = 1.0;
>   // Set the target pose
>   move_group_base_arm.setPoseTarget(target_pose);
> ```
> 
> I have tried this both with the stretch robot and the mobile base I am working with both set up with the stretch_kinematics_plugin. The planner always fails, with a message of this type. ![image](https://private-user-images.githubusercontent.com/117512744/263953821-180aac4b-54fd-43fb-99b7-f1abee7aba93.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcyNjkxNDIsIm5iZiI6MTcwNzI2ODg0MiwicGF0aCI6Ii8xMTc1MTI3NDQvMjYzOTUzODIxLTE4MGFhYzRiLTU0ZmQtNDNmYi05OWI3LWYxYWJlZTdhYmE5My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QwMTIwNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZWM1ZmQ4YTMwNzJiYTJmYTM5NDEzYTY0MmQyMTgzY2I0YzMzNjUxMjc1NTZmNDUyNDVlMjQyOTA4ZjkzNjk3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.mPpZmLqFvQyFQ5mHqs4Usk5HzQ4KFrj_y451zivlTuA)
> 
> Sorry, for the edit I posted the wrong issue initially.
> 
> ### Your environment
> * ROS Distro: [Galactic]
> * OS Version: e.g. Ubuntu 20.04
> 
> ### Expected behaviour
> Plan to a desired pose using the Cpp API.
> 
> ### Actual behaviour
> Plan fails ![image](https://private-user-images.githubusercontent.com/117512744/263953612-4dcc56eb-225a-4f84-b617-8e4b50b0a1a6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcyNjkxNDIsIm5iZiI6MTcwNzI2ODg0MiwicGF0aCI6Ii8xMTc1MTI3NDQvMjYzOTUzNjEyLTRkY2M1NmViLTIyNWEtNGY4NC1iNjE3LThlNGI1MGIwYTFhNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QwMTIwNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZWJjMGVkYzk3Mzg1YjcwZDI4Y2U0MWZhODdhODQ5ZDM1MjNhZjk4NDc2MWZhMjdhOWVlODAwMDUxNTk4NDM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.RNAhh3VNIQZziR_w9hWo6rV1aEwQibFjDnlvWKAy19w)
> 
> ## Error message
> console log