moveit/moveit2 #2337: Unable to build from source

glitchyordis, 2023-09-03:
### Description

Overview of your issue here.
Hi, I'm facing errors when building MoveIt2 from the source. Can you help me understand why? I've uploaded the terminal outputs below.

### Your environment
* ROS Distro: [Foxy]
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? Binary (I'm not sure how to check the release version, but I followed the instructions on [this page](https://moveit.ros.org/install-moveit2/source/)). 
* Which RMW (Fast DDS or Cyclone DDS)?  Cyclone

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output
[terminal outputs when building from source](https://gist.github.com/glitchyordis/a2f103f5fb6c68849b3d4a2873249854) to copy-paste the console output or segfault backtrace using gdb.

sjahr, 2023-09-04:
I think you're having an error building a workspace with the MoveIt2 tutorials :thinking: The error is in a package that is neither a part of moveit2 nor a dependency:
```bash
...
Summary: 1 package finished [5.11s]
  1 package failed: kortex_api
...
```
I am unsure what exactly went wrong but you should make sure that this package is supported by your ROS 2 version + I highly recommend using a newer version than Foxy because it is [EOL since June](https://docs.ros.org/en/rolling/Releases.html). Only  Humble and Iron are supported at this point.

sjahr, 2023-09-04:
Since this is not a bug I am closing this issue here. The right place for asking questions like this on https://robotics.stackexchange.com/. A question like this gets more visibility there

closed 2023-09-04 by sjahr