moveit/moveit2 #2352: [Servo] Using TF lookup instead of robot state for transforms causing drift

sea-bass, 2023-09-09:
I was seeing some weird drift in the new MoveIt Servo, and finally found the source of the issue.

In new servo we're doing conversions to the planning frame using a TF buffer:

```cpp
const auto planning_to_command_tf =
    transform_buffer_.lookupTransform(command.frame_id, servo_params_.planning_frame, rclcpp::Time(0));
```

In the old servo, we're doing conversions to the planning frame using the robot state:

```cpp
moveit::core::RobotStatePtr robot_state = planning_scene_monitor_->getStateMonitor()->getCurrentState();
const moveit::core::JointModelGroup* joint_model_group =
    robot_state->getJointModelGroup(servo_params_.move_group_name);
const Eigen::Isometry3d planning_to_command_tf_eigen =
   robot_state->getGlobalLinkTransform(servo_params_.planning_frame).inverse() *
   robot_state->getGlobalLinkTransform(command.frame_id);
```

While functionally this is the same, it's causing significant drift in these systems.

New servo (even after the correction in https://github.com/ros-planning/moveit2/pull/2350):
https://github.com/ros-planning/moveit2/assets/4603398/43caf2e6-bb25-42ab-a115-884b57a9c1f6

Old servo:
https://github.com/ros-planning/moveit2/assets/4603398/c4b778fe-59f3-4f12-a94a-9763fed33d49

---

So, I think we should have some kind of check on whether the frame name can be retrieved using the robot model or not, and that should decide whether to use TF.

@AndyZe @ibrahiminfinite

AndyZe, 2023-09-10:
sounds ok to me  :+1:

ibrahiminfinite, 2023-09-10:
Yeah, I agree as well.

sea-bass, 2023-09-10:
BTW the `doTransform()` method applied on an Eigen 6d vector will automatically apply the spatial twist w/ translation and rotation, so I made some tweaks on that other PR related to this.

sea-bass, 2023-09-10:
Here's what I'm thinking:

* The implementation of `RobotState::getLinkModel()` shown [here](https://github.com/ros-planning/moveit2/blob/main/moveit_core/robot_model/src/robot_model.cpp#L1345-L1346) has the code necessary to find, in O(1) time, whether a link exists in the robot model. So we should pull that out into its own function that won't log an error every time we look it up.
* So, if the link is found in the robot model, grab it directly that way; otherwise, use the TF lookup.

If this sounds good, I can work on the change.

sea-bass, 2023-09-10:
One more bug... I found the issue with TF was because the transform we were looking up was reversed... that and we were always using the spatial twist. TF works fine now.

ibrahiminfinite, 2023-09-10:
I remember fixing the reversed TF issue in the original PR itself, after Andy noticed the wrong behaviour, so the fix was the bug? LoL

sea-bass, 2023-09-10:
Might have been that the `doTransform()` function does the full spatial twist calculation instead of just the rotation portion, and flipping the transform back around maybe made this behavior less apparent?

I still think my incoming PR should be checked by you guys because at this point I've made a bunch of changes.

PR #2350 (2023-09-13): [Servo] Fix bugs when halting for collision + transforming commands to planning frame
This PR fixes a few bugs:
1. Robot was being commanded to zero state when halting for collision, now we simply don't publish outputs in this case.
2. When applying twists about the EE frame, the translation component of the transform was also being used, plus the TF `lookupTransform()` call had arguments reversed, causing drift. Fixes the issue described in https://github.com/ros-planning/moveit2/issues/2352, and applies the same fixes to pose tracking conversions as well.
4. Changed the stopped velocity thre [... cut]
closed 2023-09-13 by sea-bass