moveit/moveit2 #2363: Moving to named target with incompletly specified group joints fails to plan

2b-t, 2023-09-14:
### Description
When commanding a robot to a **named target that does not specify all the joint values** either with:
-  **[`MoveGroupInterface::setNamedTarget`](https://github.com/ros-planning/moveit2/blob/27e5d0e93a58bf4b57e201b7ed34130971a64f8d/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L1627)** or
- [`MoveGroupInterface::getNamedTargetValues`](https://github.com/ros-planning/moveit2/blob/27e5d0e93a58bf4b57e201b7ed34130971a64f8d/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L1604) followed by [`MoveGroupInterface::setJointValueTarget`](https://github.com/ros-planning/moveit2/blob/27e5d0e93a58bf4b57e201b7ed34130971a64f8d/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L1666)

the robot seems to use `0.0` for the unspecified joints and as a result might **fail to plan the motion** as it might incorrectly determine a position.
My main question is, is this expected behavior (and if yes, what is the reasoning behind it) or is this a bug?

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Source build: Branch [`galactic`, commit `1255936`](https://github.com/ros-planning/moveit2/tree/1255936a99a4e5c8625c4d6c2a4171cc5baa271a)
* RMW: CycloneDDS `0.8.0-focal.20221203.084342`

I assume the bug exists also in the other distros.

### Steps to reproduce
I have encountered an unexpected error when working with the mobile manipulator [Hello-Robot Stretch](https://github.com/hello-robot/stretch_ros2). The particularity of this robot is that its linear axis `joint_lift` at position `0.0` is in collision with its base, meaning that moving the arm to `0.0` is in general not possible (see images below).
When commanding the robot to a **named target that does not specify all the joint values** for all of its joints in the `srdf`, which is the case for most of the named targets defined in [`stretch_description.srdf`](https://github.com/hello-robot/stretch_ros2/blob/galactic/stretch_moveit2/config/stretch_description.srdf) (e.g. the `stretch_arm` group defines 7 joints
```
<group name="stretch_arm">
  <joint name="joint_lift"/>
  <joint name="joint_arm_l4"/>
  <joint name="joint_arm_l3"/>
  <joint name="joint_arm_l2"/>
  <joint name="joint_arm_l1"/>
  <joint name="joint_arm_l0"/>
  <joint name="joint_wrist_yaw"/>
</group>
```
while the group state `wrist_in` only defines a single joint, the `joint_wrist_yaw`:
```
<group_state name="wrist_in" group="stretch_arm">
  <joint name="joint_wrist_yaw" value="3.1415"/>
</group_state>
```
) the robot fails to plan the motion complaining about a **collision between the arm and the mobile base** while the two are clearly not in collision. When using RViz for planning on the other hand this works just fine!

A code snippet that is able to replicate this on the Hello-Robot stretch is the following:
```
std::string const planning_group_arm {"stretch_arm"};
std::string const named_target {"wrist_in};
auto move_group_interface_arm {std::make_shared<moveit::planning_interface::MoveGroupInterface>(node, planning_group_arm)};

if (!planning_group_arm->setNamedTarget(named_target)) {
  RCLCPP_ERROR_STREAM(this->get_logger(), "Could not find named target '" << named_target << "' for arm!");
  return false;
}
moveit::planning_interface::MoveGroupInterface::Plan plan {};
if (planning_group_arm->plan(plan) != moveit::planning_interface::MoveItErrorCode::SUCCESS) {
  RCLCPP_ERROR_STREAM(this->get_logger(), "Planning for arm for named target '" << named_target << "' failed!");
  return false;
}
if (planning_group_arm->execute(plan) != moveit::planning_interface::MoveItErrorCode::SUCCESS) {
  RCLCPP_ERROR_STREAM(this->get_logger(), "Moving arm to named target '" << named_target << "' failed!");
  return false;
}
```
replicating this might be hard as Hello-Robot is currently not offering a simulation package for the robot but one should be able to replicate this on any robot by **specifying a named target only for some joints of a group inside the `srdf`**, making sure that there is a **collision if all joints are set to `0.0`** (I assume could also be with an artificially inserted obstacle or one from an octomap from the depth camera) and then **command the robot to move to that named target with `setNamedTarget`**.

### Expected behaviour
The robot should move to the given pose either:
- keeping the other joints in the same way as in the current configuration or
- move the other joints to avoid collisions (as they are unconstrained) in case reaching the goal is not possible without that

So when commanding the robot to `wrist-in`, corresponding to setting `joint_wrist_yaw` to `3.1415 rad` or `180 deg` I would expect the robot to do the following (shadow indicates the current pose while the solid robot the desired one):

[image]

### Actual behaviour
[... 50 lines cut ...]
[rviz2-6] [INFO] [1694688854.648441388] [move_group_interface]: Replanning: no
[stretch_moveit_server-7] [INFO] [1694688858.765735270] [move_group_interface]: Ready to take commands for planning group stretch_gripper.
[stretch_moveit_server-7] [INFO] [1694688861.624239440] [move_group_interface]: Ready to take commands for planning group stretch_head.
[stretch_moveit_server-7] [INFO] [1694688864.253433426] [move_group_interface]: Ready to take commands for planning group mobile_base_arm.
[stretch_moveit_server-7] [INFO] [1694688872.271582910] [stretch_moveit_server]: Started Stretch Moveit server...
[stretch_moveit_server-7] [INFO] [1694688876.652217605] [move_group_interface]: MoveGroup action client/server ready
[move_group-5] [INFO] [1694688876.653200934] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-5] [INFO] [1694688876.653634363] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[stretch_moveit_server-7] [INFO] [1694688876.653471534] [move_group_interface]: Planning request accepted
[move_group-5] [INFO] [1694688876.664533968] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-5] [INFO] [1694688876.666819181] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'stretch_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-5] [INFO] [1694688876.790012063] [moveit_kinematic_constraints.kinematic_constraints]: Constraint satisfied:: Joint name: 'joint_lift', actual value: 0.000042, desired value: 0.000000, tolerance_above: 0.000100, tolerance_below: 0.000100
[move_group-5] [INFO] [1694688876.790053627] [moveit_kinematic_constraints.kinematic_constraints]: Constraint satisfied:: Joint name: 'joint_arm_l3', actual value: 0.000049, desired value: 0.000000, tolerance_above: 0.000100, tolerance_below: 0.000100
[move_group-5] [INFO] [1694688876.790065025] [moveit_kinematic_constraints.kinematic_constraints]: Constraint satisfied:: Joint name: 'joint_arm_l2', actual value: 0.000020, desired value: 0.000000, tolerance_above: 0.000100, tolerance_below: 0.000100
[move_group-5] [INFO] [1694688876.790074581] [moveit_kinematic_constraints.kinematic_constraints]: Constraint satisfied:: Joint name: 'joint_arm_l1', actual value: 0.000012, desired value: 0.000000, tolerance_above: 0.000100, tolerance_below: 0.000100
[move_group-5] [INFO] [1694688876.790092832] [moveit_kinematic_constraints.kinematic_constraints]: Constraint satisfied:: Joint name: 'joint_arm_l0', actual value: 0.000008, desired value: 0.000000, tolerance_above: 0.000100, tolerance_below: 0.000100
[move_group-5] [INFO] [1694688876.790135919] [moveit_kinematic_constraints.kinematic_constraints]: Constraint satisfied:: Joint name: 'joint_wrist_yaw', actual value: 3.141561, desired value: 3.141500, tolerance_above: 0.000100, tolerance_below: 0.000100
[move_group-5] [INFO] [1694688876.790346339] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'base_link' (type 'Robot link') and 'link_gripper' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[move_group-5] [INFO] [1694688876.790362945] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[... cut ...]
[move_group-5] [INFO] [1694688876.909641684] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-5] [INFO] [1694688876.909701561] [moveit_move_group_default_capabilities.move_action_capability]: No motion plan found. No execution attempted.
[stretch_moveit_server-7] [INFO] [1694688876.987217394] [move_group_interface]: Planning request aborted
[stretch_moveit_server-7] [ERROR] [1694688877.053870978] [move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
```

sea-bass, 2023-12-05:
This should try use the current state instead of falling back to zero. Will take a look.

sea-bass, 2023-12-13:
OK I just took a look and I think your initial question has it right -- so it appears to be expected behavior, but I agree that it is confusing when you haven't initialized all the joints to the current state. Namely:

> Joint variables in the group that are missing from variable_values remain unchanged (to reset all target variables to their current values in the robot use setJointValueTarget(getCurrentJointValues())).

So I think your code can follow this recommendation:

```cpp
move_group_interface->setJointValueTarget(move_group_interface->getCurrentJointValues());
move_group_interface->setNamedTarget("my_named_state");
```

Does this work?

---

Also, I have to say that Galactic is no longer supported, so if you'd like to try this on Humble/Rolling to see if the issue still persists, might be worth a shot. However, I don't know if this part of the code has really changed all that much in the last decade :)

2b-t, 2023-12-13:
_Hi @sea-bass_ 
Thanks for your response. That is quite similar to what I ended up doing. Wrote this issue some month ago when Galactic was still supported. We are using Humble now.
Feel free to close this issue if you think this default behavior should/will not be changed anytime soon.
_Many thanks_

sea-bass, 2023-12-13:
Yeah, I think because users will probably want to control exactly what to do with those other joints, I'm inclined to close this issue.

Either way, this will serve as a good reference for others running into similar problems, so thank you for posting this detailed writeup!

closed 2023-12-13 by sea-bass